navaismo/Ender-3V3-SE #1: Glitches when using the OctoPrintDetailsPageinLCD branch

RememberMe13, 2024-11-04:
Hi, when using the OctoPrintDetailsPageinLCD branch on my printer, after some time the main homepage gets overlayed over the print progress screen. i think this may be because of the screen dimming because of the delayed effects.

navaismo, 2024-11-04:
By any chance did you got a Photo of the glitch? I have printed a 5 hours job without issue. What about your job how much was idle?

Did something else happened like Paused print or trying to cancel it? 
In the meantime I'll try to reproduce it with another job of 3+ hours.

RememberMe13, 2024-11-04:
I'm just away from home at the moment, but I'll try and get a photo in a few hours.

RememberMe13, 2024-11-13:
Sorry for the long wait but here is a photo of it happening.
[image]

navaismo, 2024-11-13:
Oh, weird. Do you have any pointers to replicate the Issue? Haven't seen that in normal BAU operation.
Need to check in whih situation happens to try to locate the issue.

Also I've pushed a new version of the Octoprint Plugin try to update it and let me know, so in the meantime let me know what happened to get that screen.

navaismo, 2024-11-17:
Hi @RememberMe13 
There are new versions of the Firmware and the Plugin, please update and let me know if the issue still happens.

navaismo, 2024-11-18:
Hi @philrobbo, please let me know:

-Which version of Octoprint-plugin?
-Which version of Ender's firmware?
-How do you started the print (from Octoprint GUI or through OrcaSlicer Print menu)?
-Short description of the process.
-If possible a picture of the glitch.

Yesterday was pushed new versions of Ender's firmware (use main branch since is merged and updated) and Octoprint plugin today with some fixes(v0.0.9). Be sure to have the latest of both.

Regards.

philrobbo, 2024-11-18:
Hi Max
I am about to install the latest plug in and try.
I am using the latest ender firmware from the main branch.
I use Cura. I slice the job and press print and it sends it to Octoprint.
Phil
[image: unnamed.jpg]

philrobbo, 2024-11-18:
Max

I updated the firmware and the plug in and I still have the same problem.

Phil

navaismo, 2024-11-18:
Hi again.

So the problem is because the Octoprint plugin is configured to work with Orca files and recently fixed to work from Orca Print option.

To make it work with cura I'll need a file to analyze it, could you please generate a small cube and slice it in cura and share the file with me? Cura doesn't run on my Linux that's why I cant generate a file.

philrobbo, 2024-11-18:
is this ok?

philrobbo, 2024-11-18:
> Hi I dont see any file, can you attach it here as txt or send the link?

Hi Max
I don’t understand.
I have the file on my PC and I attached it to the email.
How do I add it as txt?
Maybe if I zip the file. I will try again soon.
Phil

Sent from Phils Mobile

philrobbo, 2024-11-18:
hi Max
can you see it now

philrobbo, 2024-11-19:
Max

Try this link

https://www.dropbox.com/scl/fi/uf8qokxtfsgtt6e82p7m8/test.zip?rlkey=9pm580u0oidruezvz3ip4swhi&st=o0ucx6i2&dl=0

philrobbo, 2024-11-19:
Or this one

https://www.dropbox.com/scl/fi/4q4sgy5s604w8ipjq1bk4/test.stl?rlkey=6x4pk8c9fwyfxokze1nkwc3iu&st=n2boy7dn&dl=0

navaismo, 2024-11-19:
Thanks, I will work on that to patch the Octoprint Plugin.

philrobbo, 2024-11-19:
Thanks mate. You’re a good man

Sent from Phils Mobile

navaismo, 2024-11-19:
From the issue https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails/issues/3  which looks like the same as above.
The  [v0.1.0 of the plugin](https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails/releases/tag/v0.1.0) has been released along with the [Firmware 1.0.9.7](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7)

Try it and let me know if that worked.

philrobbo, 2024-11-20:
> From the issue navaismo/OctoPrint-E3v3seprintjobdetails#3

Max

That works perfect.

Thank you very very much for your help. Myself and many other appreciate it.

Phil

navaismo, 2024-11-21:
Closing this, seems OP dropped this firmware and some fixes were made in the meantime.

closed 2024-11-21 by navaismo