navaismo/Ender-3V3-SE #2: Ender 3 V3 SE Screen

countrykid123123, 2024-11-05:
Hey, I'm not sure if I missed an install step or something but I have also had some issues with the progress screen only updating a layer number in the middle of the screen while printing as shown in the attached photo. When the print finishes it seems to update the whole screen with the completed screen (attached photo). Otherwise the pressure advance and filament run out sensor and other changes seem to be working properly.

[image]
[image]

navaismo, 2024-11-05:
Hi @countrykid123123 
This looks like an overlapping of the M117 messages and the O9000 cmd, you cant use both. If you want to see the progress using the M117 command with the Octoprint plugin  [DisplayLayerProgress](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress) then I suggest to use the main branch.

If you want to use the O9000 cmd from [OctoPrintJobDetails](https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails) then you must disabled the DisplayLayerProgress plugin.

The OctoprintJobDetails plugin indeed update the screen on every Z change because is the effective way to tell the printer to update the values since the printer is only listenning for updates from OctoPrint server.

navaismo, 2024-11-05:
[In this Link you can see](https://forum.creality.com/t/ender-3-v3-se-config-settings-to-roll-our-own-firmware-please/6333/135) how it works using Main Branch and DisplayLayerProgress Plugin.

[In this Link you can see](https://forum.creality.com/t/ender-3-v3-se-config-settings-to-roll-our-own-firmware-please/6333/137) how it works the OctoPrintDetailsPageinLCD branch with the OctoPrintJobDetails plugin and the DisplayLayerProgress plugin disabled.

countrykid123123, 2024-11-05:
Ok first I wanted to thankyou for the detailed response and all your work making the customized firmware and plugin. In my case I am not sure what is happening as I didn't actually have the displaylayerprogress plugin installed. I have the firmware from this fork (downloaded and installed from the firmware folder) and the companion Octoprintjobdetails for octoprint. I am running octo4a though so maybe that is causing some of the issues I am having. It is almost like it doesn't see the other events and only notices and send the command on the print completion.

navaismo, 2024-11-05:
Oh I see. Lets dig into this.
In the Octo4A do you have the terminal console? If so, could you please load the attached GCODE file(rename to .gcode)
[Cube_PETG_5m45s.txt](https://github.com/user-attachments/files/17632674/Cube_PETG_5m45s.txt), remove your filament(to not waste material) and print it. I want to see if in your console we can see the Update commands and the update responses from printer.

Also about layers numbers I would like to see if your OctoPrint setup is sending the message: "M117 DASHBOARD_LAYER_INDICATOR"

And finally, which slicer are you using and which plugins do you have installed in your Octo4A?

countrykid123123, 2024-11-06:
Ok thanks! Yes I do have the terminal console and here are the first lines 
[Test Print Terminal Start of Print.txt](https://github.com/user-attachments/files/17641114/Test.Print.Terminal.Start.of.Print.txt) and then the last lines of that cube print. [Test Print Terminal End of Print.txt](https://github.com/user-attachments/files/17641071/Test.Print.Terminal.End.of.Print.txt).
When I load the file nothing changes and no commands are sent in the terminal until I actually click print which is where that first txt doc starts. Looks like it sends the O9000 cmd to change layers and at the end but just not with any of the other events.

I am using Orcaslicer 2.2.0-beta (for printing the cube you sent me I just loaded it through Octoprint in my chrome web browser.)
Octoprint server is running 1.10.3 (it was updated today before trying the cube print). Plugins I have Installed currently are all enabled unless stated as disabled.
Achievements Plugin
Action Command Notification Support
Action Command Prompt Support
Announcement Plugin
Anonymous Usage Tracking
Application Keys Plugin
Backup & Restore
Bed Visualizer (1.1.1)
Cancel Objects (0.5.0)
Classic Webcam
Comm-fix
Core Wizard
Dashboard (1.19.12)
Discovery (DISABLED)
Error Tracking
Event Manager
File Check (2024.11.4)
Firmware Check (2021.10.11)
GCode Viewer
Logging
OctoApp (2.1.6)
OctoEverywhere! (3.6.3)
Octolapse (0.4.5) (DISABLED)
OctoPrint-E3v3seprintjobdetails (0.0.7)
Simple Filament Change Buttons (1.2)
Software Update
Top Temp (0.0.2.3)
UI Customizer (0.1.9.91)
Virtual Printer

navaismo, 2024-11-06:
Hi and thanks for the Info.
So I installed the Octo4a to make a complete debug and just realized that my plugin depends on 3 plugins.

Please to make it work correctly install the following plugins:

- Dashboard
- GcodeViewer
- PrintTimeGenius

To see correctly the layer progress you must wait till GcodeViewer analyse the file, after that you can start the print.
PrintTime Genius Provides the Estimated Print Time which is used to render the Screen in the LCD thats why it was failing the plugin to show the initial screen.

Usualy I load the file and when I see the screen in the LCD I check if gcodeviewer finished and then I press print to start the job.

So after that, try to reprint the cube without filament and let me know if is fixed.

Going to add those instructions to the readme.

countrykid123123, 2024-11-06:
Yes that has totally fixed the issue! Works great now!

navaismo, 2024-11-07:
Perfect!
Thanks for reporting the issue.

closed 2024-11-07 by navaismo