navaismo/Ender-3V3-SE #3: Leveling Failure after Immediately update from V1.0.6 to V1.0.9.5 of This fork

SHABRAWii, 2024-11-10:
### **Problem**
After updating the firmware I decided to try to print something from octoprint and then abort it on the screen.
The extruder went to home successfully and bed after hitting limit switch and tried to move it failed and shows that on octoprint and everything on the printer stuck and require restart.
I retried multiple times but still facing the same problem

> [image]

### **Assumption**
Maybe because this firmware is a fork from bed mesh 7x7 to the original 5x5 you need to redo leveling. 

I will keep you updated. 😁

SHABRAWii, 2024-11-10:
### **Solution**
I retracted the filament to get a clean bed leveling and started leveling
After that, this problem is solved. but unfortunately, I encountered a lot of problems after that in screen glitches

closed 2024-11-10 by SHABRAWii

SHABRAWii, 2024-11-10:
Update: reviewing this issue solved the glitches till now
https://github.com/navaismo/Ender-3V3-SE/issues/2#issuecomment-2456061774

navaismo, 2024-11-10:
Thanks for the feedback and solutions.