navaismo/Ender-3V3-SE #5: Updating the screen make extruder stops that cause ozzing in the part

SHABRAWii, 2024-11-10:
### Problem

> When there a long lines or low mesh files that need to send a huge number of lines fast the head stops to update the screen which causes oozing

### Suggested Solution

> I suggest increasing the buffer size and increasing the baud rate to 250000 and doing a stress testing like the square border of 210mmx210mm in a 250mm/s to make sure that this won't be a problem

navaismo, 2024-11-10:
Thanks for the heads up, will check that. If you have a gcode file to use and reproduce the issue please attach it.

navaismo, 2024-11-20:
I'm testing  the recommendations of increasing the Baudrate to 250000 and also double the buffersize in Marlin. But the problem is that I'm having a lot of Checksum issues comming back from the printer like:

``` bash
2024-11-19 20:10:05,177 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 53157, current line = 53158
| Last lines in terminal:
| Send: N53149 G1 X117.173 Y109.863 E.24177*88
| Recv: ok
| Send: N53150 G1 X117.173 Y110.083 E.00691*87
| Recv: ok
| Send: N53151 G1 X117.294 Y110.438 E.01181*95
| Recv: ok
| Send: N53152 G1 X117.478 Y110.594 E.00761*86
| Recv: ok
| Send: N53153 G1 X117.753 Y110.663 E.00893*84
| Recv: ok
| Send: N53154 G1 X122.238 Y110.663 E.14119*83
| Recv: ok
| Send: N53155 G1 X122.522 Y110.59 E.00922*100
| Recv: ok
| Send: N53156 G1 X122.758 Y110.342 E.01076*88
| Recv: ok
| Send: N53157 G1 X122.819 Y110.083 E.00839*95
| Recv:  T:222.14 /222.00 B:71.97 /72.00 @:73 B@:37 FAN0@:99
| Recv: Error:checksum mismatch, Last Line: 53156
| Recv: Resend: 53157
```
Sometimes even the prints stop for few seconds till it recover the missing point.

I'm thinking if could be a Hardware issue related to bad cable or interference or maybe UnderVoltage from RPi; Will change cable and try from OCTO4A.

Also wondering if is too much for the STM updating every Z change+time change... Kind of lost.
Does the Plugin&FW affects so much the prints? I'm making basic printing without issues but nothing that needs accurate scales.

navaismo, 2024-11-21:
Still having issues with a new cable and OCTO4A, looks like the SMT can handle those speeds but I'm going to ask to the experts in the Marlin discussion forum first.

navaismo, 2024-11-22:
Published a possible fix with the recommendations.
BaudRate increased to 128000 and double the buffers.

Feel free to reopen the issue if needed.

closed 2024-11-22 by navaismo