navaismo/Ender-3V3-SE #10: Print TIme Remaining

rtorchia, 2024-11-25:
Is it possible to use the M73 command embedded in gcode to get estimate time remaining, instead of using PrintGenius?

navaismo, 2024-11-26:
I guess its possible. Can you elaborate or share a link to understand how it works? My Gcodes has no M73 embeded so need to understand how it is generated to enable it in the octoprint plugin.

rtorchia, 2024-11-26:
I got the idea from the Octoprint plugin Slicer Estimator at https://github.com/NilsRo/OctoPrint-SlicerEstimator

It also exposed an API for other plugins to use

navaismo, 2024-12-24:
I was checking in OrcaSlicer the option to Enable the M73 command and the result is like this:

```bash
$ cat M73Cube_PLA_5m2s.gcode | grep M73
M73 P0 R5
M73 P1 R4
M73 P3 R4
M73 P4 R4
M73 P5 R4
M73 P6 R4
M73 P7 R4
M73 P8 R4
M73 P9 R4
M73 P10 R4
M73 P11 R4
M73 P12 R4
M73 P13 R4
M73 P14 R4
M73 P15 R4
M73 P16 R4
M73 P17 R4
M73 P18 R4
M73 P19 R4
M73 P20 R4
M73 P20 R3
M73 P21 R3
M73 P22 R3
M73 P23 R3
M73 P24 R3
M73 P25 R3
M73 P26 R3
M73 P27 R3
M73 P28 R3
M73 P29 R3
M73 P30 R3
M73 P31 R3
M73 P32 R3
M73 P33 R3
M73 P34 R3
M73 P35 R3
M73 P36 R3
M73 P37 R3
M73 P38 R3
M73 P39 R3
M73 P40 R3
M73 P40 R2
M73 P41 R2
M73 P42 R2
M73 P43 R2
M73 P44 R2
M73 P45 R2
M73 P46 R2
M73 P47 R2
M73 P48 R2
M73 P49 R2
M73 P50 R2
M73 P51 R2
M73 P52 R2
M73 P53 R2
M73 P54 R2
M73 P55 R2
M73 P56 R2
M73 P57 R2
M73 P58 R2
M73 P59 R2
M73 P60 R2
M73 P60 R1
M73 P61 R1
M73 P62 R1
M73 P63 R1
M73 P64 R1
M73 P65 R1
M73 P66 R1
M73 P67 R1
M73 P68 R1
M73 P69 R1
M73 P70 R1
M73 P71 R1
M73 P72 R1
M73 P73 R1
M73 P74 R1
M73 P75 R1
M73 P76 R1
M73 P77 R1
M73 P78 R1
M73 P79 R1
M73 P80 R1
M73 P80 R0
M73 P81 R0
M73 P82 R0
M73 P83 R0
M73 P84 R0
M73 P85 R0
M73 P86 R0
M73 P87 R0
M73 P88 R0
M73 P89 R0
M73 P90 R0
M73 P91 R0
M73 P92 R0
M73 P93 R0
M73 P94 R0
M73 P95 R0
M73 P96 R0
M73 P97 R0
M73 P98 R0
M73 P100 R0
```

I can edit the plugin to add this option just the time left  when less than an hour(60 mins) will be like above just the reported minutes without seconds, is that ok?
I was looking into the other plugin but couldnt see an API that expose the progress or time, I saw it exposes changes of filament, and it says it reads the same M73 from gcode.

rtorchia, 2024-12-24:
I was just checking in to github to send you info on this, and how orcaslicer adds M73 codes unless you disable it here:

[image]

Anyways, this would be and awesome addition, as the M73 command reports time (minutes) and percentage remaining:

M73 P<percent> [R<minutes>]

navaismo, 2024-12-24:
Great, will work on this and let you know the version to test.

navaismo, 2024-12-24:
@rtorchia plugin version is ready to test: [v0.1.3](https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails/releases/tag/v0.1.3)

Let me know if works as expected.

rtorchia, 2024-12-28:
Busy holidays, so I just had time today to try this out. 
It is working great! The Layer, Time Left and Percentage circle all are updated on the display correctly. The only issue is the Print Time is at zero. This should be initially set to the first Time Left value, as apposed to be left at zero.

Awesome work! Loving that the display is back as a useful device with this printer

navaismo, 2024-12-28:
HI, weird the Print time should be taken from the analisis as usual, can you try with this file or send me yours?
[M73Cube_PLA_5m2s.txt](https://github.com/user-attachments/files/18266759/M73Cube_PLA_5m2s.txt)

rtorchia, 2024-12-28:
I used your file and same thing, here is what it looks like during printing:

[image]

Looking at OctoApp on my phone, gives me this info from Octoprint:

[image]

navaismo, 2024-12-28:
Is it possible to collect the octoprint.log file? I'm seeing the time with the cubic file.

rtorchia, 2024-12-28:
Okay, I've cut down the logfile to the time it started printing to when I canceled it halfway:

```

2024-12-28 13:39:59,067 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: M73Cube_PLA_5m2s.gcode, owner: torchia, user: torchia
2024-12-28 13:39:59,079 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: FileSelected
2024-12-28 13:39:59,097 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2024-12-28 13:39:59,101 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: M73Cube_PLA_5m2s.gcode, owner: torchia, user: torchia
2024-12-28 13:39:59,117 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2024-12-28 13:39:59,125 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2024-12-28 13:39:59,149 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2024-12-28 13:39:59,281 - octoprint.plugins.action_command_notification - INFO - Got a notification: Bed Heating...
2024-12-28 13:39:59,580 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> Loaded File, Waiting for PrintStarted
2024-12-28 13:39:59,613 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: PrinterStateChanged
2024-12-28 13:39:59,718 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: PrintStarted
2024-12-28 13:39:59,719 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Plugin Version: 0.0.1.3
2024-12-28 13:39:59,719 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Sliders values:
2024-12-28 13:39:59,722 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Enable O9000 Commands: True
2024-12-28 13:39:59,723 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Progress based on Layer: False
2024-12-28 13:39:59,728 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Progress based on M73 Command: True
2024-12-28 13:39:59,729 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>+++ PrintStarted with Loaded File waiting for metadata
2024-12-28 13:39:59,746 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2024-12-28 13:40:00,235 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Getting Print Details Info 1st time.
2024-12-28 13:40:01,049 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2024-12-28 13:40:01,136 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - File selected: /octoprint/octoprint/uploads/M73Cube_PLA_5m2s.gcode
2024-12-28 13:40:01,139 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Total layers found: 50
2024-12-28 13:40:01,139 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Print time still unknown
2024-12-28 13:40:01,141 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>+++ PrintStarted with M73 command enabled
2024-12-28 13:40:01,147 - octoprint.plugins.SlicerEstimator - INFO - Send Metadata Print Event for file M73Cube_PLA_5m2s.gcode
2024-12-28 13:40:01,150 - octoprint.plugins.dashboard - INFO - Gcode not pre-processed by Dashboard. Processing now.
2024-12-28 13:40:01,153 - octoprint.plugins.dashboard - INFO - GcodePreProcessor started processing.
2024-12-28 13:40:01,212 - octoprint.plugins.dashboard - INFO - Gcode pre-processing done.
2024-12-28 13:40:01,212 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found layers: 51
2024-12-28 13:40:01,212 - octoprint.plugins.dashboard - INFO - GcodePreProcessor found filament changes: 0
2024-12-28 13:40:01,212 - octoprint.plugins.dashboard - INFO - GcodePreProcessor saving layer count in file metadata
2024-12-28 13:40:01,267 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: ChartMarked
2024-12-28 13:40:01,280 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: GcodeScriptBeforePrintStartedRunning
2024-12-28 13:40:01,292 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: GcodeScriptBeforePrintStartedFinished
2024-12-28 13:40:01,308 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: PrinterStateChanged
2024-12-28 13:40:01,343 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: Home
2024-12-28 13:40:01,358 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: plugin_SlicerEstimator_metadata_print
2024-12-28 13:40:08,795 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: SettingsUpdated
2024-12-28 13:40:28,874 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.181
2024-12-28 13:40:28,877 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: ClientClosed
2024-12-28 13:40:29,934 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: PositionUpdate
2024-12-28 13:40:29,968 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: PositionUpdate
[... cut ...]
2024-12-28 13:45:20,992 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Event detected: ZChange
2024-12-28 13:45:20,993 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Update Print Details Info
2024-12-28 13:45:21,078 - octoprint.util.comm - INFO - Cancelling on request of the printer...
2024-12-28 13:45:21,081 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2024-12-28 13:45:21,087 - octoprint.util.comm - INFO - Force-sending M108 to the printer

```

navaismo, 2024-12-28:
Thanks, I can see that the analysis wasn't complete at the start and then when forcing the updates time wasn't populated.

Let me check on that and I'll get back to you to see if the bug is solved.

navaismo, 2024-12-29:
Ok looks like when you disable the Print Time Genius plugin, and there is no other analysis made the Print time goes to 0, Fixed my octoprint plugin to get and set the print time from M73 command.

Please try the [v0.1.4](https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails/releases/tag/v0.1.4) of Octoprint plugin and let me know if now works as expected.

rtorchia, 2024-12-29:
Working as expected now! Very nice! Thank-you!

closed 2024-12-30 by navaismo