navaismo/Ender-3V3-SE #9: Can't connect Octoprint to Printer in v1.0.9.7_3

findleybc, 2024-11-25:
Serial communication times out in Octoprint with v1.0.9.7_3. I think it is a problem with 128000 baudrate. It works connecting at 115200 baud using v1.0.9.7. I suspect that is it s problem with Octo4a not supporting 128000 baud but.

Octo4a Version 2.1 (latest)
Octoprint Version 1.10.3
OctoPrint-E3V3SEPrintJobDetails (0.0.1.1)

Terminal screen when trying to connect:
_Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer_

navaismo, 2024-11-25:
You need to enable the baudrate in the Octoprint serial settings in order to select 128000

findleybc, 2024-11-25:
> You need to enable the baudrate in the Octoprint serial settings in order to select 128000

Thanks for your reply. I've done that. 

_Connecting to port /dev/ttyOcto4a, baudrate 128000
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x7bc2897d60, open=True>(port='/dev/ttyOcto4a', baudrate=128000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello_

navaismo, 2024-11-25:
Probably as you estated could brand issue with Octo4A, I use a raspberry without issues. You can downgrade to the previous version the _3 version it's only to increase the connection and avoid oozing if present.

closed 2024-11-26 by navaismo

navaismo, 2024-11-27:
Looks like it is an issue with the implementation of Octo4a --> https://github.com/feelfreelinux/octo4a/issues/347