navaismo/Ender-3V3-SE #11: Mesh data display on LCD only partially functional

jaaxx, 2024-12-02:
After initiating leveling from the LCD I receive the completed message which I assume is normal behavior.  However, when I goto Edit screen all cells show 0.  If I then select a cell to edit and click, the value for that cell and that cell only is displayed.  So I would have to Edit/Select/Click each cell in order to view the mesh.  The mesh is in fact valid and functions correctly, but it is terribly cumbersome.

Is this a known or expected behavior?  I can initiate a mesh from Bed Visualizer as usual (with some minor visual issues due to 5x5 grid)

navaismo, 2024-12-04:
Before editing the values did you saved the mesh by clicking in the store configuration option menu?

About the GUI yes I'll work on that to make it decent, but si far everytime I re-level the bed valued are stored correctly.

If possible let me see your serial log.

navaismo, 2024-12-04:
Can you check if this is related to your issue:
 https://github.com/navaismo/Ender-3V3-SE/issues/12#issuecomment-2515965878

navaismo, 2024-12-16:
New release has been published [v1.0.9.7_4](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_4)
This will solve the overlaping messages allowing to save the configuration after bed level finish if started from LCD menu.

navaismo, 2024-12-24:
No feedback received, closing for now

closed 2024-12-24 by navaismo