navaismo/Ender-3V3-SE #12: Failed to enable Bed Leveling

mikey0000, 2024-12-04:
Bed leveling either via Octoprint or the display fail

gets part way through then errors out, I reset the configuration and that didn't help either, once it fails the fan ramps up. Printer is completely unresponsive.

mikey0000, 2024-12-04:
Update: running it with octoprint disconnected seems to have solved it + a full reset. Not sure why having Octoprint connected is a problem, maybe its that invalid mesh comes through as an error and octoprint calls M112 ?

navaismo, 2024-12-04:
This is strange, I have leveled my bed many times with that version.

Can you attach the serial log when trying from Octoprint?

mikey0000, 2024-12-04:
Running now, even if I run bed leveling directly with octo detatched it doesn't work properly even though it says it has completed. It basically doesn't complete properly, running with logs now.

mikey0000, 2024-12-04:
[serial.log](https://github.com/user-attachments/files/18001277/serial.log)
attached log, though this time it didn't error out, but its still not saving the bed leveling, still all zeros

navaismo, 2024-12-04:
Did you saved the configuration by selecting the Store configuration option menu?

Anyway I'm AFP(away from printer) so will check and fix if needed when I have access to it again.

Thanks for reporting.

mikey0000, 2024-12-04:
ok.... it seems turning it on and off again multiple times has resolved the issue... as for the display, it shows when I do an edit, running again using the display to see if it does it again.

mikey0000, 2024-12-04:
I think I may have worked it out. It was throwing a mesh error, I think due to the grid size change for leveling the existing saved configuration didn't match, throwing the printer off. So fix is to save configuration again then run leveling. This should be reproducible going from stock to this. After hitting the save configuration, it stopped errorring out.

closed 2024-12-04 by mikey0000