navaismo/Ender-3V3-SE #13: Store configuration doesn't work

mikey0000, 2024-12-06:
Not sure whats happening now, but every time I turn the printer on and off it goes back to language selection and calibration.

Any ideas, possibly linked to my previous issue. Happy to provide logs or whatever you need me to do to help diagnose this.

mikey0000, 2024-12-06:
I should add, works great otherwise.

mikey0000, 2024-12-06:
Do you have the eeprom file on your sd card?
Yeah it seems to be resetting every time it boots as it does seem to be saving. I'm going to try and re-flash it again. wiped the sdcard but I believe the configuration is stored directly on the machine on this version?

navaismo, 2024-12-07:
No, I dont use SDCard anymore y flash the machine and then remove the sd crd, thats why is quite odd that your printer is rebooting and reseting.

mikey0000, 2024-12-07:
Sdcard with or without it still resets, I'm going to blame the firmware, perhaps once the latest marlin code is applied it will resolve it.

navaismo, 2024-12-16:
New release has been published [v1.0.9.7_4](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_4)
This will solve the overlaping messages allowing to save the configuration after bed level finish.

About the reset has nothing t do with this firmware that behaviour is not normal and no others reported that. Routines for save and store are factory defaults.

mikey0000, 2024-12-17:
Well flashing this version has now resolved the issue. :man_shrugging: maybe was just a bad flash.

closed 2024-12-17 by mikey0000