navaismo/Ender-3V3-SE #14: Changing filament with M600 command no longer resumes print.

zxUltra, 2024-12-07:
It will do everything as it's supposed to, including lifting the Z axis more, but after changing the filament, it just hangs. Pressing the pause button just extrudes a bit of filament, then does a whole lot of nothing. The M600 command has always been a dumpster fire on this printer, like how it cuts power to the heat block unless you press the screen button, it sounding like a time bomb, with the print seeming to resume whenever the printer feels like it. Now it doesn't feel like doing it at all. 

Note: I am not using Octoprint on this printer, this is from the SD card.

artsitemail, 2024-12-07:
The prints is sending the comand m600 to octoprint even without it.
Its the same where, its only works with octoprint connect to the printer.
I've already talked about this on the forum with the firmware creator and he'll look into it.

navaismo, 2024-12-07:
Hi there!
Yes, this is in my list to check whats going on. Unfortunately can't debug and make firmware changes till Dec 16th so bear with me.

navaismo, 2024-12-24:
Closing due to no response from reporter

closed 2024-12-24 by navaismo

zxUltra, 2024-12-30:
Hello, sorry for not replying in time. I'm happy to report that the issue is indeed fixed. Thank you so much for your efforts.