navaismo/Ender-3V3-SE #15: E3v3seprintjobdetails problem

Rareshacks, 2024-12-22:
Hello,

I am writing to you because I am experiencing some issues with the E3v3seprintjobdetails plugin.

The first issue is that when the print starts, it refreshes itself but then it stops updating the layer information, time, and the progress percentage of the print job.

When I pause the print from the plugin, upon resuming, the printer performs an uncontrolled movement before it resumes printing.
[image]

navaismo, 2024-12-23:
HI, I'm seeing that the total lyers are not parsed correctly.

- From which slicer was produced?
- How did the Job was loaded, from Slicer or Uploading to Octoprint?
- Can you share the octoprint.log?
- Whihc progress are you using, Time progress or Layer progress?

About the movements after job is resumed depends on how you have configured in octoprint, in the plugin's github there are some examples.

Rareshacks, 2024-12-23:
Hello, I’m following up with a response regarding the issue I’m facing:

As a slicer, I use Cura.
I upload the G-code to OctoPrint. Initially, the plugin worked within the parameters, but after a while, it started returning "none" for the layer after printing the guide line. It shows the current layer, but the print status is not uploaded. The plugin is set to display the layer. At the end, after the print finishes, it shows that the G-code is 100% loaded, and it asks for confirmation. But this happens after the part has been printed.
I use Layer Progress.
Regarding the uncontrolled movements, I’ve used your examples, but I’m facing a small issue. It extrudes too much filament upon resume, which clogs the nozzle due to the excess. I’ve adjusted the codes to prevent this, but uncontrolled movements on one of the axes still happen during the resume phase, and then it prints normally. Here are my examples:

Pause Script:
{% if pause_position.x is not none %}
; ===== Pause Script =====
; Set relative positioning
G91
; Raise Z by 10 mm for safety
G1 Z10 F1200
; Set absolute positioning
G90
; Move to a safe position on X and Y
G1 X10 Y10 F6000
; Retract filament by a safe amount
G91
G1 E-3 F600 ; Moderate retraction of 3 mm
G90
; Display a message on the screen
M117 Print Paused
M300 ; Beep for pause
{% endif %}

Resume script:
{% if pause_position.x is not none %}
; ===== Resume Script =====
; Set relative positioning
M83
; Reintroduce filament by a controlled amount
G1 E3 F600 ; Introduce only 3 mm of filament
; Set absolute positioning
M82
; Reset absolute XYZ positioning
G90
; Return to the previous print position
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F6000
; Reset feed rate if present
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
; Display a message on the screen
M117 Resuming Print
M300 ; Beep for resume
{% endif %}

[octoprint (2).log](https://github.com/user-attachments/files/18233705/octoprint.2.log)

navaismo, 2024-12-24:
About movements [here is a video on how it pauses with the configuration ](https://i.imgur.com/I8Bxehq.mp4)of the github in Octoprint. So not sure which other confs are you setting or which movements are you seen, if you check there is no purge during pause or resuming just normal Oozing of filament.

About layers seems your files arent saved in the upload folder of your setup. Based on your log:

> 2024-12-22 11:41:34,235 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - got a path: 
> /home/rares/.octoprint/uploads/123gcode.gcode
> 2024-12-22 11:41:34,239 - octoprint.plugins.E3V3SEPrintJobDetails - ERROR - Error reading file /home/rares/.octoprint/uploads/123gcode.gcode: [Errno 2] No such file or directory: '/home/rares/.octoprint/uploads/123gcode.gcode'
> 2024-12-22 11:41:34,240 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Layers found: None
> 2024-12-22 11:41:34,241 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - File selected: 123gcode.gcode
> 2024-12-22 11:41:34,241 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Total Layers: None
> 2024-12-22 11:41:34,242 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Current Layer: 0
> 2024-12-22 11:41:34,242 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - Progress: 0
> 2024-12-22 11:41:34,852 - octoprint.plugins.E3V3SEPrintJobDetails - INFO - >>>>>> E3v3seprintjobdetailsPlugin Update Print Details Info

Its a good practie to disabel DisplayLayerProgress plugin if you are going to use this one.
And if your setup is not standard  then better change to Time Progress because if the plugin cannot parse the file the progress will not be updated since total layers is none.

Tested with Octoprint plugin 0.1.2 & Printer v1.9.7_4

Rareshacks, 2024-12-26:
With the latest update, it seems to be working well, thank you.

Rareshacks, 2024-12-26:
Regarding the uncontrolled movements, I have attached 2 videos in which the uncontrolled movement can be observed at the moment when I give summaries from the pctoprint. The movement occurs when the pause is given from the plugin interface on the printer. The filament stepper does not immediately start with the print, it pours a few layers and then it also starts to push the filament. Thank you

https://github.com/user-attachments/assets/7222d231-bb7e-40c3-960d-a857977bac99

https://github.com/user-attachments/assets/d7b08964-04dd-4efd-a5b3-1dee534441b1

navaismo, 2024-12-26:
The movements are due to the setup of Octoprint, you can see in my previous video it does the same like a fast wipe. 
If you are not comfortable with the setup of the scripts you can overrride it with your own to match your desired movements.

For example just raise the nozzle:
```
G91 Z              ; Set to relative positioning mode
G1 Z15             ; Move hotend away from part by 15mm
G90 Z              ; Set to absolute positioning mode
```
and then get it back:
```
G91 Z              ; Set to relative positioning mode
G1 Z-15             ; Move hotend away from part by 15mm
G90 Z              ; Set to absolute positioning mode
```
(Dont use the above is just and example )

About the extrusion issue when pausing from knob's printer I couldn't reproduce it. My printer went back and printed normally the rest of the object. But as I commented in the latest release since the FW is getting modified to work with Octoprint Host if you want to use the LCD do not connect Octoprint, The latest FW was modified to have a separate flow when Octo is connected and when is not and may have weird responses when mixing commands.

I will double check later this week the pause-extrusion issue when using LCD, I just need to finish some prints to test it thoroughly.

navaismo, 2025-01-30:
Closing due to inactivity.