navaismo/Ender-3V3-SE #16: Question regarding baud rate and Octoprint

thallesgate, 2025-01-02:
Is it possible to increase it even more? I've detected some printing quality issues when using octoprint, and i believe it has to do with USB communication bottleneck.  (See [https://www.reddit.com/r/ender3/comments/bnqkoq/a_follow_up_to_my_zitsblobs_solution_tips_post/](url) and [https://community.octoprint.org/t/noticeably-worse-quality-from-octoprint-than-from-sd-card-ender-3-v2/33797](url) - both are from ender 3 v2 but i'm having related issues on my v3 se)

I'm installing platform.io to do some testing.

navaismo, 2025-01-02:
I have tested  with higher speeds https://github.com/navaismo/Ender-3V3-SE/issues/5

But Connection become unreliable, dropping a lot of packages and causing bad effects on prints. Tried with different hardware and cables but seems the Board can't handle above 128000.

thallesgate, 2025-01-07:
I was able to replicate your results and have even more printing issues by raising it to 250000 baud. You're totally right! 
I will investigate further my RPi. If i find anything i will re-open this issue. Thank you so much!

closed 2025-01-07 by thallesgate