navaismo/Ender-3V3-SE #21: Show "Your bed level is OK" after bed leveling

eduard-sukharev, 2025-01-20:
It's a pretty frequent refrain in V3 SE subreddit that people ask whether their bed levelling is ok or not and how to fix that.

What I propose is after bed levelling is done, and if bed leveling is within ok-ish margin, show an affirmative message "Your bed leveling OK!". Probably even vary the score - PERFECT, OK, MEDIOCRE, BAD. For MEDIOCRE and BAD results show the QR code for video on printer setup and adjusting the Y-Z skew (the one that is shown on bed leveling error).

Both the affirmative message and QR code should be shown in a Popup dialog above the leveling screen and should be hidden once user accepts that.

[image]

This should reduce the user confusion. Optionally, make this dialog accessible via something like ❓ icon and add additional QR code to wiki page explaining results.

eduard-sukharev, 2025-01-20:
I should probably start this discussion on subreddit and prepare a wiki there or anywhere else to at least have a set of advices for novices.

navaismo, 2025-01-20:
This is a good idea. 
Just some thoughts about the UI and the HMI control of the buttons:
- The grid and values need to be saved before showing the popup. Because the popup will overwrite the screen and the OK button needs an HMI control from the knob, so I think it will return to the grid.
I was showing a simple message after completion and pressing the back button erased the grid and values so people reported that was not saved the new ones. Removed the new message solved the issue.

- About the QR did you manage to upload new images or planning to use the QR icon that already exists on the memory of the LCD?

About the discussion can you join us in this [forum thread](https://forum.creality.com/t/ender-3-v3-se-config-settings-to-roll-our-own-firmware-please/6333?page=7) please? There is were everything started and most of the features that this firmware has, has been discussed and suggested there.

eduard-sukharev, 2025-01-20:
Yes, implementing the dialog should be done only with preventive save. That same dialog should also probably be shown in Edit Bed Leveling dialog too.

Regarding the QR code - I haven't yet tried to connect USB-to-SERIAL to the screen for quick QR tests, so I kinda yet hope it works. If it's not, then either show a shortened link or no link at all.

navaismo, 2025-01-20:
> Regarding the QR code - I haven't yet tried to connect USB-to-SERIAL to the screen for quick QR tests, so I kinda yet hope it works. If it's not, then either show a shortened link or no link at all.

I tried directly from printer and didn't worked but havent tried with the USB-Serial adapter yet.

navaismo, 2025-01-22:
Privet Eduard.

Yes, painful path to discover the LCD instructions since seems very different from Manufacterer to the code we have. I was going to buy  the same LCD model from manufacturer but [instructions set](http://wiki.tjc1688.com/debug/index.html) made me to stop the purchase.  

Im interested about decompile it, havent tried yet if you can share some insigths I can take a look.

eduard-sukharev, 2025-01-23:
Well, there's a decompiled version of DWIN_SET resources,  namely `9.ICO` file but only for Ender3V2S1 screens, and you already familiar with those, but there's none for our E3V3SE printer, which  makes things harder - i'm not even sure what you are going to decompile then. I have made a script to cycle through all icons in screen and while most of the icons are the same, they might be placed there by different indexes and we still have more icon assets (like those qr codes). So yeah, extracting those icons in full would be a good starting point for creating custom assets.

eduard-sukharev, 2025-02-01:
Hey, I've managed to add QR code rendering routine - the smallest version of qr, only 25 alphanumeric characters, but should be enough for short links and being shown on screen (no need for high error correction levels).
It uses QRCodeGenerator library from platformio packages registry and is actually pretty nice to use.

[image]

navaismo, 2025-02-01:
Nice!

navaismo, 2025-02-10:
Added your function in the latest release: https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_8 
Thank you very much for your time and help.

closed 2025-02-10 by navaismo

eduard-sukharev, 2025-02-10:
This should probably be reopened until this featyre is actually implemented - the one you merged only adds support for custom QR codes, but no actual hints and support links were added.

afiskon, 2025-02-11:
Привет, Эдуард.

As long as this feature doesn't intervene with the way the firmware works right now (all the heights are visible to the user, etc) it would be nice to have.

Thanks for working on this!

eduard-sukharev, 2025-02-12:
This feature, when implemented, should add additional popup dialog that shows bed leveling assesment. When that popup dialog is dismissed, user should be back to the bed leveling values shown to him.

eduard-sukharev, 2025-03-01:
Hola, Max.

Can we add a wiki to this repo and start filling it with basic useful things to be later added in such UI dialogs (with QR codes or directly with url shorteners)?

Reddit is a good place for discussion and quick help, but since we provide custom firmware build we could also provide with a good FAQ and basic printing knowledge.

navaismo, 2025-03-01:
For sure, we can do that.

navaismo, 2025-03-01:
I have created just a dummy page and it supposed to be open to collaborators so chack if you can edit it.

eduard-sukharev, 2025-03-01:
I successfully pulled it and committed, but it says I can't push as i don't have permissions.

navaismo, 2025-03-02:
Check if you received the access/invite

navaismo, 2025-06-20:
The latest merge solves this one?

PR #52 (2025-06-19): Bed leveling assessment popup
This feature adds a helpful popup window after bed leveling is complete to help users assess bed leveling results.

resolves #21