navaismo/Ender-3V3-SE #27: Several feature requests

afiskon, 2025-02-11:
Hi,

Firstly, thank you a lot for maintaining this fork. I recently purchased my E3v3SE and was extremely happy to learn that I can have linear advance and input shaping.

There are several features that IMO would be nice to have. I don't know if they are difficult to implement or if anyone will be interested in doing so, but I figured it will not hurt to share my thoughts on this.

**Feature 1.** The ability to turn off the beeper. I don't like the fact that the screen beeps on my and having an ability to disable it would be nice.

**Feature 2.** Better procedure of filament replacement. Personally I prefer the "push" method when I cut the filament on top of the extruder and push it down with a new filament. Unlike the "pull" method, it ensures that little pieces of plastic can't stuck in the throat, which happened to me in the past. 

At first I tried to heat up the nozzle and move E coordinate manually. I quickly discovered that Marlin crashes when E reaches 256 mm [to clarify, I tested this only on the stock firmware]. Then I tried to use the Extrude menu. It works OK, however I should use it twice in order to push down enough filament. Before first and second use E3v3SE parks the head and heats up the nozzle, which takes some unnecessary time. Also, the temperature is constant 240 degrees, which will not work for any filament type.

What I would like to see ideally is the ability to set the nozzle temperature I need and then have several buttons, like Extrude / Retract and 5 / 10 / 25 / 100 mm. I've seen this in Flying Bear Ghost 6 firmware which also runs Marlin.

I hope you will find this little piece of feedback useful. If something above doesn't make sense and/or needs clarification, please don't hesitate to ask.

navaismo, 2025-02-11:
здравствуйте!

> **Feature 1.** The ability to turn off the beeper. I don't like the fact that the screen beeps on me and having an ability to disable it would be nice to have.

You mean totally disable it? Or Only when you press the knob. I think the E3V3SE has two beepers one in the LCD Knob, and the other in the board to beep on several macros. For example: M600 when you need to pay attention to the filament change.

> **Feature 2.** Better procedure of filament replacement.

Didn't get this. Currently the printer has in the Prepare Menu the options. _**Extrude & Retract**_ that works to change the filament when no jobs are present.

When you press the lever at the side of the extruder hole you can insert the filament till the end and then Press the _**EXTRUDE**_ option this will purge the filament and will take care of any other residual color on it.

To change the filament you press the option **_RETRACT_** it heats the fliament, extrude a little and then starts moving the gears backwards to take out the filament. In this you can press the lever again and pull fast or just wait till the gears ejec the filament.

I found those options working good so far after a year.

afiskon, 2025-02-12:
> You mean totally disable it? Or Only when you press the knob. I think the E3V3SE has two beepers one in the LCD Knob, and the other in the board to beep on several macros. For example: M600 when you need to pay attention to the filament change.

I meant only when I press the knob. This is a minor issue of course.

> Didn't get this. Currently the printer has in the Prepare Menu the options. Extrude & Retract that works to change the filament when no jobs are present. [...]

Yes, but there are two problems:

1. You can't change the temperature. The default 240 works OK for PLA and PETG but may not work for other materials. The modified firmware allows heating up the nozzle up to 300 degrees which allows printing with PA12 (nylon) for instance with an appropriate hotend.

2. EXTRUDE menu works OK but personally I wonted to extrude a little more filament in order to purge the nozzle appropriately. In my use case I have to use EXTRUDE twice which works but is a bit inconvenient. Note that I don't use RETRACT at all (I cut the filament and push it down with another one) because I had problems with this before. One in a while a little piece of cold filament may stuck in the hotend throat and block the new filament. For this reason I need to EXTRACT twice as much filament as it does by default, and there is no option to change the extracted length.

Does this answer your questions?

guilhermelemes, 2025-02-12:
I agree that disabling the beep when pressing the knob would be a nice modification. 
Also, the ability to turn off the LCD screen during print (on the Tune in-print menu).

Both of these features are present on the firmware I installed on my Ender 3 S1 (among several other modifications). 

I don't know how different the firmwares between S1 and V3 SE are, but if there's any similarity that could help applying these features, here goes mrisco's Ender 3 S1 firmware link: 
https://github.com/mriscoc/Ender3V2S1

navaismo, 2025-02-12:
> Both of these features are present on the firmware I installed on my Ender 3 S1 (among several other modifications).

Whats the string in the LCD Menu to disable the buzzer and LCD? So I can take a look on the mriscoc git.

guilhermelemes, 2025-02-12:
> [@guilhermelemes](https://github.com/guilhermelemes)
> 
> > Both of these features are present on the firmware I installed on my Ender 3 S1 (among several other modifications).
> 
> Whats the string in the LCD Menu to disable the buzzer and LCD? So I can take a look on the mriscoc git.

"Enable sound"
"Turn Off LCD"
"LCD Brightness" (another good one)

They are both in "Advanced" menu and in-print "Tune" menu

navaismo, 2025-02-12:
@guilhermelemes 
Thanks! 
I'll take a look on that and see if can be ported to this.

@afiskon 

> 2\. EXTRUDE menu works OK but personally I wonted to extrude a little more filament in order to purge the nozzle appropriately. In my use case I have to use EXTRUDE twice which works but is a bit [...]

Whats the idea that you have in Mind? Another Menu that look like:
Manual Extrude (if pressed)
    Set Temp(degrees)                 0-300
    Extrude Lenght(mm)              5-20 
    Start

After start pressed, heat nozzle till input temp, when reached extrude input lenght?

That will take some time to implement, but let me know about the menu UI & controls.

afiskon, 2025-02-12:
@navaismo Here is how it's done in Flying Bear S1 for instance:

[image]
[image]
[image]

The "Auto" / "Manual" switch on the left is rather useless, so instead of "Manual" / 5mm / 10mm / 20mm I would prefer 5 mm / 10 mm / 25 mm / 100 mm. But other than that this interface is perfect.

Hopefully this gives you an idea of the proffered UI. Naturally when adopted for E3v3SE it will look differently. The important things here are the ability to set the nozzle temperature, the length of the extraction / retraction, and extract / retract buttons.

navaismo, 2025-02-21:
The [Release v1.0.9.7_9](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9) now includes the feature to mute and unmute the LCD Knob . 

The option is under Prepare Menu, last row.

[image]

About the Extra Extrusion feature is going to take a while, due to other projects.

navaismo, 2025-02-24:
Just a FYI that I'm stalled due to some time consuming prints(3 days) so the custom extrude menu is still under development.

navaismo, 2025-03-01:
@afiskon 
In the latest [Release v1.0.9.7_9b](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9b) I have added the custom extrusion menu.

Sorry for the GUI but cannot do much with the limited graphic resources and screen of the V3SE, so its kind old but works.

#### Usage:
> [!TIP]
>
> 1. Set your desired temperature above 195°C, else an error will come out.
> 2. Set your desired extrusion length above 10mm, else an error will come out.
> 3. Click on proceed.
> 
> After that the Noozzle will heat till +-3 Degrees of the desired temp and will extrude the desired length automatically.
>
> Finally just click on the confirm button to go out the menu.
>

<div align="center">

[image]

</div>

If you are agree we can close this issue/feature request.

Dan263737, 2025-03-02:
Can you please add automatic step motors calibration setting? If you add this we will have better, detailed an smooth motions of all axes. I mean something like auto steps per mm option. Thank you

navaismo, 2025-03-02:
> Can you please add automatic step motors calibration setting? If you add this we will have better, detailed an smooth motions of all axes. I mean something like auto steps per mm option. Thank you

Please open a new Issue requesting this, and provide a background or similar feature to be based on.

afiskon, 2025-03-03:
@navaismo thank you a lot! The screenshots look great. I will test the new release this week, probably today.

afiskon, 2025-03-03:
Custom extrude works great, thank you a lot! ❤

closed 2025-03-03 by afiskon

navaismo, 2025-05-02:
Just a FYI, the position of this feature is now moved to the left to avoid the extruded filament  touching the bed.