navaismo/Ender-3V3-SE #30: Linear Advance seems not to work

putoale, 2025-02-14:
Hello,

I flashed the 1.0.9.7_7 version and I've spent the last days printing Pressure Advance calibration tests, but the results seem to be the same regardless of the selected K-factor. I'm sure I'm properly setting up the K-factor through M900 commands, and if I send M900 from the Octoprint terminal it answers with the correct value. The gcode of the objects I'm printing contains the M900 code at the beginning. However it doesn't seem to have any effects on the print result. As recommended by queeup I've disabled all the other options in orca slicer that could interfere with linear advance. I've also disabled power-off recovery and z height fade to fix the issue with the linear advance stopping working for h>10mm. 

Comparing your firmware with the one from queeup, I noticed that #define S_CURVE_ACCELERATION is commented in his firmware, and it's not commented in yours. On the internet, some people say that you cannot activate both LINEAR_ADVANCE and S_CURVE_ACCELERATION, so maybe that's why I don't see any changes.

I'll try to build the firmware myself with this change, but I just wanted to know if I'm the only one having this issue.

Thanks :)

navaismo, 2025-02-14:
I do not use linear advance on my prints, and for this fork you need to activate from Slicer:
Refer to this comment: https://forum.creality.com/t/ender-3-v3-se-config-settings-to-roll-our-own-firmware-please/6333/278

navaismo, 2025-02-14:
btw that's not the latest release, this one is: https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_8b

putoale, 2025-02-14:
> I do not use linear advance on my prints, and for this fork you need to activate from Slicer: Refer to this comment: https://forum.creality.com/t/ender-3-v3-se-config-settings-to-roll-our-own-firmware-please/6333/278

Oh, ok thanks. I've already activated it in the slicer, but it doesn't work. I guess that I'll have to disable the S_CURVE_ACCELERATION and try again. I will post here the results if you think this may be useful for someone else.

> btw that's not the latest release, this one is: https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_8b

I'll download the latest version, thanks :)

navaismo, 2025-02-14:
> will post here the results if you think this may be useful for someone else.

Sure thing, will be very helpful for everyone. 
Thx.

putoale, 2025-02-15:
> I can't confirm this bug report. I'm using OrcaSlicer and LA works just fine when enabled in the slicer settings.

What calibration method did you use? What’s the final K-factor you used? I’ve tried both with PA line, PA pattern and PA tower in a range 0 to 0.2 and I honestly don’t see any differences in the print quality. I’m using SUNLU PETG right now, but I’ll also try with PLA

afiskon, 2025-02-15:
I used eSUN PLA and OrcaSlicer tower method https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#Tower-method (Note that **ideally** input shaping should be disabled with `M593 F0` in the start G-code. The documentation doesn't say that. This however doesn't has a huge effect, just makes the test more accurate). 

Note that this test requires the linear speeds to be high. I used 100 mm/s. Also make sure you disable "Slow down for better cooling" option and verify that the tower is printed at 100 mm/s.

I ended up with 0.1 LA factor for PLA.

[image]
[image]
[image]
[image]

putoale, 2025-02-16:
> Note that this test requires linear speeds to be high. I used 100 mm/s. Also make sure you disable "Slow down for better cooling" option and verify that the tower is printed at 100 mm/s.

I found out that you can't change the walls speed in orca slicer for this test, I think it's hard coded in the gcode and it will be 100mm/s regardless of the selected speed. However I've disabled the "Slow down for better cooling" option and I've printed it for a higher range of K-factors. Nothing changed.

Trying with the PA pattern test from 0.1 to 0.3 the difference is still minimal. I've got the impression that for K>0.2 there are some gaps near the angles, so I eventually selected 0.2. However looking at the results people are getting with this test, usually they seem to have more noticeable results with even smaller ranges.

PA line test from 0.05 to 0.3 also doesn't show any big differences.

I guess I'll try it again with PLA and check if it's related to PETG.

navaismo, 2025-02-16:
@rtorchia  you use PA right?
Can you confirm or help here?

rtorchia, 2025-02-17:
Yes, I do use, and did the calibrations for, PA. I used PLA for the calibration prints, and did see visible differences, however subtle, to do my calibrations on the SE. For most PLA I'm using a PA of 0.096.

I kept all the default settings for the calibration tests, and didn't change anything. For the filament, I chose a generic Creality PLA. Once I completed the tests and examined the results, I turned on PA for the filament with the determined value.

putoale, 2025-02-18:
> Yes, I do use, and did the calibrations for, PA. I used PLA for the calibration prints, and did see visible differences, however subtle, to do my calibrations on the SE. For most PLA I'm using a PA [...]

I tried again with PLA, and now the difference is visible (though difficult to notice). So I guess it was more related to the filament than to the firmware

navaismo, 2025-02-19:
Can we close the issue?

putoale, 2025-02-19:
> Can we close the issue?

Yes, Thanks for helping 🙂

closed 2025-02-19 by putoale