navaismo/Ender-3V3-SE #31: Failed to enable Bed Leveling

Kabron287, 2025-02-16:
FW: FW10978b_OctoPrintPageinLCD_115kbps.bin
Everything OK until I made manual Leveling from LCD panel.
Printer hangs on Auto Z offset in progress.  
Octoprint sends several warnings:
Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Failed to enable Bed Leveling

There was an error reported by your printer
Your printer's firmware reported an error:

Failed to enable Bed Leveling

Due to this error, OctoPrint was not able to continue to communicate with your printer and sent an emergency stop command M112 to ensure all heaters are turned off, then disconnected.

Terminal:
Recv:  T:50.64 /0.00 B:43.42 /0.00 @:0 B@:0 FAN0@:0
Recv: X:134.25 Y:125.00 Z:10.00 E:0.00 Count X:10740 Y:10000 Z:4000
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"

Kabron287, 2025-02-16:
Without USB connection to Octoprint manual Leveling performs OK.
also Printer Display now has a larger font and does not fit messages in window.

[image]

navaismo, 2025-02-16:
> Due to this error, OctoPrint was not able to continue to communicate with your printer and sent an emergency stop command M112 to ensure all heaters are turned off, then disconnected.

This is expected you need to disconnect Octoprint and do a new level since the mesh is changed.

> Without USB connection to Octoprint manual Leveling performs OK.
> also Printer Display now has a larger font and does not fit messages in window.

Thats not my plugin.
Thats DisplayLayer Progress, which should be disabled if you are going to use the O9000 cmds from https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails 
You can configure DLP  from it settings page.

closed 2025-02-16 by navaismo