navaismo/Ender-3V3-SE #35: Incorrect estimated print time

eduard-sukharev, 2025-02-21:
Some gcode files are shows with incorrect estimated print time. Example faulty gcode files are attached.

This needs investigation and fix

[incorrect_print_time.tar.gz](https://github.com/user-attachments/files/18907732/incorrect_print_time.tar.gz)

navaismo, 2025-02-21:
- [  ]  If using Octoprint:
Depends directly on which Plugin make the estimation there are several like Prin Time Genius or Estimator Calculator. If none is installed then Octoprint use the Bytes of the Gcode to make the calculation and I want to say is horrible not accurate. When starting to make the Octoprint Plugin discover it is a known issue.
Thats why I recommend to use M73 to get progress and Time, based on the Gcode file itself.

- [  ] If printing from SD: 
I saw the code it use the same approach as reading bytes and is also a Known issue for example [this post](https://www.reddit.com/r/Ender3V3SE/comments/1is0w9k/well_thats_a_new_one_also_i_do_not_believe_you/) 
What we can do is enable the M73 to use the progress for the SD Card print too.

What do you think?

eduard-sukharev, 2025-02-21:
That post from reddit is where I got those files from - i'm `Kraplax` there

eduard-sukharev, 2025-02-21:
It probably already uses the M73, I don't know, haven't checked the gcode files myself yet, but my guess is we have some issues when reading/interpreting it.

eduard-sukharev, 2025-02-21:
Ok, seems like there's no M73 in both of the gcodes provided and from what I could get from reddit threads, the `CE3V3SE_F1+calendar+2025.gcode` was estimated in Cura to be around 15 hours to print. There's a comments header section that has
```gcode
;TIME:55847
```
which, if interpreted as seconds, should be around `15h30m`. So we should check whether we parse that and how we treat that value.

navaismo, 2025-02-21:
For the SD Card Progres uses the bytes of the filesize here is the part of code that calculates the Percentage:

```C++
 static inline uint8_t percentDone() {
    if (flag.sdprintdone) return 100;
    if (isFileOpen() && filesize) return sdpos / ((filesize + 99) / 100);
    return 0;
  }
```
And rendering:
```C++
  // print process
    const uint8_t card_pct = card.percentDone();
    // Card percent=card.percent done();
    static uint8_t last_cardpercentValue = 101;
    if (last_cardpercentValue != card_pct)
    {
      // print percent
      last_cardpercentValue = card_pct;
      if (card_pct)
      {
        _card_percent = card_pct;
        Draw_Print_ProgressBar();
      }
    }
```

For the time remianing:
```C++
// Estimate remaining time every 20 seconds
    static millis_t next_remain_time_update = 0;
    if (_card_percent >= 1 && ELAPSED(ms, next_remain_time_update) && !HMI_flag.heat_flag) // Rock 20210922
    {
      // _remain_time = (elapsed.value -dwin_heat_time) /(_card_percent *0.01f) -(elapsed.value -dwin_heat_time);
      card_Index = card.getIndex();
      // card_Index = 1;
      // rock_20211115 Solve the problem of occasionally abnormal display of remaining time >100H
      if (card_Index > 0)
      {
        _remain_time = ((elapsed.value - dwin_heat_time) * ((float)card.getFileSize() / card_Index)) - (elapsed.value - dwin_heat_time);
      }
      else
      {
        _remain_time = 0;
      }
      next_remain_time_update += DWIN_REMAIN_TIME_UPDATE_INTERVAL;
      Draw_Print_ProgressRemain();
    }
    else if (_card_percent <= 1 && ELAPSED(ms, next_remain_time_update))
    {
      // rock_20210831 solves the problem that the remaining time is not cleared.
      _remain_time = 0;
      Draw_Print_ProgressRemain();
    }
```
So basycally is from filesize too and the conversion has a poor calculation like Ocotoprint without PTG plugin. Thats why the time is so random.
For the SD CARD there is no M73 parsing

eduard-sukharev, 2025-02-21:
I think we can do better. I guess upstream Marlin already does that - we could just borrow their solution. Or write our own. Mind if I get this one to myself?

navaismo, 2025-02-21:
Orca at footer section has 
```bash
 ; estimated printing time (normal mode) = 6m 8s
```

> I think we can do better. I guess upstream Marlin already does that

Haven't seen the changes yet, could be a solution.

> Mind if I get this one to myself?

Of course be my guest.

eduard-sukharev, 2025-02-21:
Does octoprint send the estimated time in any way? I could only solve for SD card printing via parsing the header comments section. I'll look into the Orca footer too. Anyways, I'll post the PR when ready.

navaismo, 2025-02-21:
> Does octoprint send the estimated time in any way?

Yes, my plugin send the M73 from gcode or the estimation from PTG current job.

> Anyways, I'll post the PR when ready.

Sure and thanks

eduard-sukharev, 2025-02-22:
ok, first of there seem to be definitely a bug - the last fragment of code you've provided contains comment that it's there to solve the exact same bug it's putting there. The _cardIndex variable is not the current position in a file, it's actually a current position on SD card, i.e. the index number of selected file. This using it in this context makes totally no sense.
Upstream Marlin has this formula correct (and it matches the commented out formula in that same code snippet). Gotta fix that, but I still cannot reproduce the bug to make sure I have fixed it.

eduard-sukharev, 2025-02-22:
ok, besides the estimated print time being calculated only by the file size or set by M73, the code for parsing the gcode file header to get the print time is actually quite closely related to parsing the thumbnail image. I can tweak a few things, but the way the thumbnail is embedded by mriscoc's script differs a bit from the way it's done by Creality slicer. Nothing critical, but will probably require more samples and testing. In the meantime, does any of your gcode files show estimations etc when you select file from SD before you start print? or does it always show 0-0-0? I know you probably not using SD printing, but if you can check - that'd be great.

navaismo, 2025-02-22:
> the code for parsing the gcode file header to get the print time is actually quite closely related to parsing the thumbnail image.

Yes, they have added even the B64 library and they have prepared everything; the issue is not to get the bytes from B64 the problem is to copy the image into a memory address like we tried to. I was also exploring the posibilty of make the imagemap from Marlin but that consume a lot of memory so I desisted and just doing by Python transmission.

There is a comment inside the creality code that says the V3SE doesnt support the decode and render and they desisted too, looks more likely the LCD is not prepared for that. SO they even bother.

> In the meantime, does any of your gcode files show estimations etc when you select file from SD before you start print? or does it always show 0-0-0? I know you probably not using SD printing, but if you can check - that'd be great.

If i recall that was working only when I was using creality print software, once I switched to Cura and then Orca never show anything again.

eduard-sukharev, 2025-02-22:
> they have prepared everything; the issue is not to get the bytes from B64 the problem is to copy the image into a memory address like we tried to.

Yep, I remember that. It's just that with thumbnail parsing routines they've cut out the whole header parsing, which also may contain `TIME` as total remaining time estimated by slicer.

I guess I'll have to enable that whole path just to get that info and we could probably figure something out for image rendering. I still hope we'll be able to do that.

navaismo, 2025-02-22:
>they've cut out the whole header parsing

They have the routine to get those details too. It in the   _Marlin/src/lcd/dwin/e3v2/lcd_rts.cpp_ at line 214
```C++
uint8_t read_gcode_model_information(void)
```

> I still hope we'll be able to do that.

Me too, Im really tempted to buy the Generic model to see the features and limitations.

navaismo, 2025-02-25:
Hi @eduard-sukharev!  I've merged your PR https://github.com/navaismo/Ender-3V3-SE/pull/39 thanks a lot for the work.
I will work on the other feature request during the week and if all goes good by the weekend release the Version with your fix, the Linear advance Menu and the custom feature to extrude.

Shall we close this?

eduard-sukharev, 2025-02-25:
We should probably do some minor testing just in case anything goes wrong and there's a division by zero errors or any other rendering/performance issues. Should not be the case, but still. Should we rather call for testers?

navaismo, 2025-02-25:
Yes I will do some tests too, and I agree that woukd be good to calling for testers that use the SD.

eduard-sukharev, 2025-02-25:
I've missed a line a merged PR and now build fails. I'll create new PR with a fix, currently here are the binaries for 115k baudrate (not sure how you build 128k)

[FW1097_9a_pt_OctoPrintPageinLCD_115Kbps.bin.zip](https://github.com/user-attachments/files/18968913/FW1097_9a_pt_OctoPrintPageinLCD_115Kbps.bin.zip)

navaismo, 2025-02-25:
> not sure how you build 128k

For 128 and 150 I just change the Baudrate in Configuration.h line 118
```C++
#define BAUDRATE 150000
```

navaismo, 2025-02-25:
Btw and a little bit off topic, Are your email and name in github set corectly? I cannot see your user in the contributions list although we have merged your PRs into the main branch.

eduard-sukharev, 2025-02-25:
hmm... not sure. I develop on my work laptop and it probably has work SSH keys. Gotta connect them to my account, I guess, or at least configure to my public email

eduard-sukharev, 2025-02-25:
I guess it's the Vigilant mode - unless I didn't sign my commits, it showed them as Unverified and that probably lead to me missing from the list

navaismo, 2025-03-01:
Since no complains and I have tested too your PR with the fix of estimated print and print details wit no issues at all, I've published it in the latest [Release v1.0.9.7_9b](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9b)

<div align="center">

[image]

</div>

Let me know if we can close this and thanks a lot for your amazing work and time.

eduard-sukharev, 2025-03-01:
yep, let's close it. awesome!

closed 2025-03-01 by navaismo