navaismo/Ender-3V3-SE #38: Can't get out from M600

Dimik4545, 2025-02-25:
Hi! When I print, when M600 command arrives, I changed filament, clicked continue print, but printer simply extruded filament and nothing, but printer has been stayed on the pause! When I click it again, nothing happens too. Are there any solutions? Thank you.

navaismo, 2025-02-25:
From the Readme:

>When using standalone printer, the Knob will control the Pause and Continue processes. If you connect the host(Octoprint) then the Knob became useless and you need to control the Pause and Continue from the Octoprint GUI until you disconnect or restart the printer.

afiskon, 2025-02-25:
M600 is also a common command for changing filament during the print. It can be manually inserted into G-code in a slicer. This is commonly used to print business cards, paintings, etc. See for instance:

- https://eax.me/files/2025/01/3d-printed-business-card.jpg
- https://eax.me/files/2025/01/3d-printed-painting.jpg

@Dimik4545 is this the use case you are struggling with? Did you try using M25 instead? https://marlinfw.org/docs/gcode/M025.html

navaismo, 2025-02-25:
When the firmware has host commands enabled there is a bug that makes the knob useless as described above, so I needed to modify the firmware a bit to make it work when the host like Octoprint is not connected and when is it. Was one of the first things reported here and fixed. That's why the behavior in the readme is described.

Dimik4545, 2025-02-25:
@afiskon @navaismo I have no Octoprint + I tried M25 command too. When M25 command arrives, happens the same, but without filament retraction. Extruder simply extrudes and nothing happens.

navaismo, 2025-02-25:
Which version of the firmware you are using?

navaismo, 2025-02-25:
I've just tested with the latest without connecting octoprint and is working as expected. 
Here is the video(some difficulties while trying with one hand xD)

https://github.com/user-attachments/assets/3a4cfe35-433c-4c1d-93a8-b138c69a4977

Dimik4545, 2025-02-25:
@navaismo latest. This is 1.0.9.7_9 I think. I will take a try to multicolor print today. I will send a result then.

navaismo, 2025-02-25:
Should be working as in the video. But I recommend to make some small tests like with a little cube or test without filament before going to the main print so you can ensure everything is ok.

Dimik4545, 2025-02-26:
@navaismo This time it's worked, but first time I clicked "continue" button it has extruded a little filament and when I clicked "continue" button again, and printer comes in the last point it stopped, retracted filament and continued print. I don't know why, but I took the filament in extruder and print gone well.

navaismo, 2025-02-26:
Hi,

One click is to tell the printer is ok to continue after parking the head, then the other click is to tell the printer to continue to print after you changed the filament.
TheMacro does extrude first and retract the current filament, then after the second click it extrudes a bit more the new filament to ensure no color ghosting, it wait a little to remove that final extrusion and then moves to continue the print job.

Regarding the 100% progress, Elapsed time and ETA is the stock firmware, it has a very poor calculation of it. 
Eduard has pushed a PR to solve this and is looking for testers https://github.com/navaismo/Ender-3V3-SE/issues/35

I personally dont use anymore the SD, since I use octoprint; so I trust Eduard on this fix.

Dimik4545, 2025-02-26:
@navaismo Спасибо! Thank you for help! I think, we can mark this issue as solved.

closed 2025-02-26 by navaismo