navaismo/Ender-3V3-SE #41: Incorrect files selected for printing

eduard-sukharev, 2025-03-02:
As discussed on [this thread](https://www.reddit.com/r/Ender3V3SE/s/f6DihOZBBR) on Reddit, due to the DOS filenames handling inconsistent it shows one file during file selection but starts different file for actual printing. I have experienced this myself and just got used to delete files before uploading for next print.

This needs fixing for people who use SD card printing.

navaismo, 2025-03-02:
привет!

Is this related with our fork or comes from stock firmware?
Before moving from octoprint I was using even a lot of folders in the SD and never happened to me.
Will check how is generating the list of files.

eduard-sukharev, 2025-03-02:
its stock creality behaviour. i had this issue before our work on parsing files etc.

lmorg, 2025-06-24:
Yeah definitely a problem that originated from stock. I've ran into it a few times myself on stock.

Problem is if you have a few files similarly named.