navaismo/Ender-3V3-SE #42: [FR] One click print

eduard-sukharev, 2025-03-02:
Marlin allows workflow that is close to the so called "[One click print](https://help.prusa3d.com/article/one-click-print_220249)" - once SD card is inserted the print start dialog pops up with last updated file on SD card selected. Should be doable and will nicely add to the UX.

navaismo, 2025-03-02:
If I understood well, we will need to:

1. Check when the SD is inserted (and in case of using Octoprint don't).
2. Check which file is the most recent. At root level I guess.
3. Popup a Message asking if want to print?
4. If so Load the Print Information Screen and then user can decide if print or not?

eduard-sukharev, 2025-03-02:
yes. the upstream marlin can do this with basic 12864 screens combining two or three defines in Configuration.h and Configuration_adv.h. It has something to do with last updated sorting, offer print on insert and probably something else.

eduard-sukharev, 2025-03-02:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1750
Guys at marlin discord were helpful to point exactly to the define to enable.

navaismo, 2025-11-23:
I think we are going to drop this for this version, lets see if in Marlin 2.1 we can enable without costing us a liver

closed 2025-11-23 by navaismo