navaismo/Ender-3V3-SE #43: Steps per mm - Step motors auto calibration

Dan263737, 2025-03-02:
Can you please add automatic step motors calibration setting? If you add this we will have better, detailed an smooth motions of all axes. I mean something like auto steps per mm Trinamic StallGuard, which allows step loss detection and compensation.  Perhaps something similar could be implemented to automatically calibrate steps to mm..

Dan263737, 2025-03-03:
Yes, I know but i think automatic calibration via the endstops bcs my printer has a very bad steps per mm setting and I cant get it to the right value.

eduard-sukharev, 2025-03-03:
My question was an alternative type of question that included 3 options to chose, giving "yes" as an answer does not provide me with any info.
Please, could you expand more one the way you see it working? Did you see such feature in any other printer? What are your expected steps to do this and what should happen afterwards?

eduard-sukharev, 2025-03-03:
If we're talking about **Adjusting the StallGuard current limit** to make marlin detect filament jams, carriage collisions and so on - it could help in detecting the nozzle clog or it could replace endstop switches (in quite inprecise way). Replacing already connected and stock endstops with imprecise StallGuard thing is a bad idea. Detecting a nozzle clog might be a viable option, but it has nothing to do with any calibration whatsoever

eduard-sukharev, 2025-03-03:
If we're talking about **Adjust StealthChop to SpreadCycle threshold** - that menu option makes sense, although defaults work pretty much fine. If we're going for maximum configurability, then that menu option would be a good one, but again, it has nothing to do with any calibration process.

navaismo, 2025-03-06:
@Dan263737  what happened? 
We were waiting for a clarification to understand what can be done. now I see it marked as complete. Is that ok?

Dan263737, 2025-03-06:
Yeah, I meant adjust StealthChop to SpreadCycle threshold for maximum configurability. I closed it bct i thought it was a bad idea.

eduard-sukharev, 2025-03-06:
not a bad one, but definitely not of the main priority, probably.

Dimik4545, 2025-03-06:
@eduard-sukharev Can we become clog detection without a sensor then, because marlin can detect it or I am not understanding it?

eduard-sukharev, 2025-03-06:
we probably could make something like that, but that seem to be something that upstream marlin can benefit from or has already implemented. And if not - we should at least check to know why. This sounds like a very unreliable way to detect nozzle clog, but better than nothing. And what should we do once we detect a nozzle clog? stop the print? beep the buzzer and go one? lower the speed?

Dan263737, 2025-03-06:
> pravděpodobně bychom mohli něco takového vyrobit, ale zdá se, že to je něco, z čeho může marlin proti proudu těžit nebo to již implementoval. A pokud ne – měli bychom si to alespoň ověřit, abychom věděli proč. Zní to jako velmi nespolehlivý způsob, jak zjistit ucpání trysky, ale lepší než nic. A co bychom měli dělat, jakmile zjistíme ucpání trysky? zastavit tisk? pípnout bzučák a jít jeden? snížit rychlost?

I think beep the beeper and pause the print😊

navaismo, 2025-03-10:
Sorry but I'm still dont know what I should implement ;D Can anyone share docs or somthing? Kinda lost here.

eduard-sukharev, 2025-03-10:
Ok, so there's a StallGuard thing in TMC drivers that can detect that the motor has hit something. Motors run at some more or less average current, but if linear rails are dirty and there's a friction the motor needs more current to overcome that friction. Drivers sense that and provide that current but they also provide Marlin with feedback about it (somehow). There's an option to remove actual physical endstops and rely solely on the fact that when print head hits the gantry limit it will sense the current consumption spike that is beyond some preset threshold and will trigger the X/Y-endstop.

In similar fashion, we could implement a current sensing threshold for E-motor and allow user to adjust it. Once the motor struggles to push the filament through then the current will raise and if it goes beyond the (properly adjusted) current limit we could trigger the error/warning.

This sounds like it could have already been implemented in Marlin or, vice versa, declined from implementation for some reason.

Dan263737, 2025-03-11:
> Omlouvám se, ale stále nevím, co bych měl implementovat ;D Může někdo sdílet dokumenty nebo něco? Nějak se tu ztratil.

I think I am lost too.

closed 2025-05-02 by Dan263737