navaismo/Ender-3V3-SE #44: Time remaining

Cordelinho, 2025-03-02:
On printer display shows >100h left
while the print itself is close to 2h 

[image]
[image]

eduard-sukharev, 2025-03-02:
Hi. Which firmware version do you use and which slicer?

Dimik4545, 2025-03-03:
@eduard-sukharev Hi! I have the same issue, this is latest firmware, this is 1.0.9.7_9b. In versions before  all was perfect.

eduard-sukharev, 2025-03-03:
> @eduard-sukharev Hi! I have the same issue, this is latest firmware, this is 1.0.9.7_9b. In versions before  all was perfect.

Hi, could you verify that exact same gcode file shows this bug in 1.0.9.7_9b but works well in 1.0.9.7_9? what slicer is this? currently OrcaSlicer will have old algorithm (pretty much wild guess based on file size).
Attach gcode file as zip to this issue.

Dimik4545, 2025-03-03:
@eduard-sukharev Okay, I will try it right now. I will send a results in +- 10-25 minutes.

Dimik4545, 2025-03-03:
@eduard-sukharev I did it. This happens even on small prints, that can take from 10 - 20 mins. 
Tried with a small 20x20mm cube:

[image]

Printer profile:

[image]

Print time:

[image]
 
Note: Maybe is it an bug or not, but on new version of firmware(1.0.9.7_9b) print doesn't start properly. It stays "Paused" but when you hover the "print" button, print starts.
Nothing special, setting "Disable set remaining print time" is enabled (Ver: 1.0.9.7_9b):

[image]

100H>;

Created another gcode file with "Disable set remaining print time" disabled:

[image]

...And got the same bug:

[image]

Downgraded to version 1.0.9.7_9 or 1.0.9.7_9a:

Same gcodes, different results:

Gcode with "Disable set remaining print time" enabled:

[image]

...and there is no problem, like on 1.0.9.7_9b, that print don't start automatically.
I think it has no sense to upload a version with "Disable set remaining print time" disabled.

eduard-sukharev, 2025-03-03:
oh wow, I didn't know that there's such an option in orca. will have to fiddle with it, thanks for extensive report, i'll have a look at it, hopefully this week.

Dimik4545, 2025-03-03:
@eduard-sukharev Are you actually an `Kraplax` from reddit?

eduard-sukharev, 2025-03-03:
yes, there and on some other platforms I'm referred to as Kraplax

navaismo, 2025-03-03:
Disable print option is basically the M73 command.
It is disabled in the source code, although I use it for the Octoprint plugin/screen.

Cordelinho, 2025-03-04:
i didn't got to check this issue until now, pretty much @Dimik4545 explained all the problems i'm having, except for me the print starts automatically after pre heating

navaismo, 2025-03-04:
Should  we enable M73 for the SD Card process?, Progress and ETA will be updated by gcode. I personally use it in Octoprint because is consistent.

eduard-sukharev, 2025-03-04:
is it not enabled by default for sd card? it is parsed as gcode and it does update the internal remaining print time, iirc

navaismo, 2025-03-04:
It is disabled in the GCODE suite.

Dimik4545, 2025-03-05:
@navaismo An idea: add an option, that enables or disables M73 in _PREPARE_ menu or simply enable it in firmware. You can do a test firmware to test, if it works. I can test it.

navaismo, 2025-03-05:
The gcode suite for M73 will need modifications to set and show the values in the display. 

@eduard-sukharev will you check the M73 assignment too or just the fix of the parsing file?

closed 2025-03-07 by navaismo

Dan263737, 2025-04-03:
Have same problem in Prusaslicer

eduard-sukharev, 2025-04-03:
> Have same problem in Prusaslicer

can you provide the gcode file you have issues with? which firmware version are you using?

Dan263737, 2025-04-04:
> > Máte stejný problém v Prusaslicer
> 
> můžete poskytnout soubor gcode, se kterým máte problémy? jakou verzi firmware používáš?
All gcodes, firmware: v1.0.9.7_9d. I also tried using creality slicer but still have <100H. Do you want specific gcode?

eduard-sukharev, 2025-04-04:
yes please, any specific file would do. Also, do you get the wrong reading at the very start of the print or some time after?