navaismo/Ender-3V3-SE #46: Some(2.5) Ideas to add

Dimik4545, 2025-03-03:
`Hi! I have an request to add some features to make this firmware more special!`
- **Adaptive bed mesh leveling**_(when possible)_: When you take something to print, printer is leveling bed mesh only on a part, where 3d model is placed. I'd like to have this feature, because when you turn the "calibration"(in before-print menu) on, then printer calibrates all of the bed and this takes many time + because of this feature we'll become more accurate prints more precisely first layer.

- **An option to edit stock screen brightness**_(too, when possible)_: When you want to take a photo, this will be hard to take a photo of it, for example, for issue or showcase, or when you go to sleep and the screen is too bright. Yeah, there are screen covers, but they are looking not so good(it's for me). You can simply change the screen brightness for example long prints, because you doesn't need it. It can be from 100% brightness, to for example, 15%. 

- **Disable/enable hotend fan**_(can be excluded)_: To turn on or turn off small hotend fan or a possibility to control the hotend fan speed. This fan is pretty loud and you can make it more slower, that this fan got more quieter. I think, that this idea was dumb, because this fan is needed to cooldown the hotend and because of this it can't be controlled. I just leave it like that. Maybe I'm right and maybe I'm wrong.
```
Спасибо!!
Thank you!!
```

navaismo, 2025-03-03:
Здравствуйте!

* Regarding the bed level, if you saved your configuration in the eeprom the mesh is available for all prints, so to use it on every print without leveling again you can just add in the slicer GCode start the command to use the mesh saved.

If I recall is M420 S1, with that you should avoid the calibration step everytime.

* For the brightness, yes is something we can do.

* For the fan is something that we need to check, haven't seen if we can control it from code yet.

eduard-sukharev, 2025-03-03:
The bed leveling off toggle does not disable the bed leveling itself, but rather does not repeat the whole process before print, also effectively re-calibrating z offset (to probably wrong value).

eduard-sukharev, 2025-03-04:
A menu option to control the dimmed lcd brightness would be great - the full brightness seem to be reasonable, but once idle timeout is passed the lcd screen can be dimmed to different states - some would even prefer to total blackout.

Eitschman, 2025-03-04:
Hi

For the hotend fan there is a variable in configuration_adv.h named "EXTRUDER_AUTO_FAN_SPEED".
255 = full speed.
I guess that it should be possible for you guys to create a menu item to set this...

Regards
Andreas

guilhermelemes, 2025-03-04:
@navaismo remember mrisco firmware, for Ender 3 S1? It has all these 3 features. 
I'm on vacation right now so I can't check LCD text for these options so that you can look for it on the code.  But searching for "LCD brightness" and other key words should work.

navaismo, 2025-03-04:
Well, the firmware for the LCD that they use is different from our LCD, last time the beeper configuration didn't match ours. But since the LCD of the V3 SE does dimm I know we have a working instruction set.

navaismo, 2025-03-04:
So back to the topic I think we should not touch the Hotend fan right? I mean that could lead into heat creeps or changing the original PID? Maybe Im just wrong.

For the LCD brightness yes is something possible just wondering if we add it as a **_TUNE_** setting or leave it in _**PREPARE**_ menu and when it dimms will do it to the preset value.

How do you see it working?

eduard-sukharev, 2025-03-04:
I think this belongs to Settings - you don't want to alter it during the print, just set and forget.
The hotend fan might be usable, but I guess it's not in any priority - one should rather get proper fans and only then turn it down if it was too high rpm.

navaismo, 2025-03-08:
Added the Display Settings Menu under prepare to control the buzzer, and brigthness of the LCD in the latest [Release v1.0.9.7_9c](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9c). For the other two I guess i will not work on it as discussed.

The bright settings are only set during runtime because I think someone by accident can put the LCD to 0 and then lost visibility :| just a precaution.
Let me know if we can close the issue.

[image]

Eitschman, 2025-03-08:
For the brightness setting: can you define a minvalue, so that the user cannot set it to 0?

navaismo, 2025-03-08:
I left the 0 in case someone want to turn off completely, then to recover just reset the printer.

Eitschman, 2025-03-08:
Hmm, ok. I this really a use case? In my opinion it makes no sense to turn off the display completely. When I have to power off and on the printer to reset the LCD, every running print will possibly be destroyed...
But I think that is a question of personal preferences. I think that saving the settings will give more benefits rather than having to adjust it after every power off.

navaismo, 2025-03-08:
Now comes to my head put the minimum value to 3 in normal bright and for the dimm allow to 0 them moving the knob will restore the brightness.

This will be applied next release then.

Eitschman, 2025-03-08:
Right, that will work. Please save the settings then!

Eitschman, 2025-03-08:
Another idea for the LCD: it could be usefull to adjust the time before the dimm is applied.

navaismo, 2025-03-09:
Should be Solved with the [Latest Release v1.0.9.7_9d](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9d)

[image]

closed 2025-03-09 by navaismo

Eitschman, 2025-03-09:
Looks very good! Thank you!

AromaticArea3836, 2025-03-09:
DUUUDE!! It's update is very cool!!!! Many new things happened in few days I wasn't watching in github!! Thank you guys all!