navaismo/Ender-3V3-SE #48: Print stats

Dan263737, 2025-03-05:
Hi. I will like if you add something like a print stats. What it shoult contain: Total print time (hours), Filament (meters), X,Z,Y axis (km), Number of prints and number of completed prints. I see this on a lots of printers so it will be good to see it on this printer. Thanks

Dan263737, 2025-03-05:
[image] Prusa MINI

Dimik4545, 2025-03-05:
Oh, really cool idea!!

navaismo, 2025-03-06:
Lets see what can be done is under review now.

navaismo, 2025-03-08:
Added the Printer Statistics Menu under Control Menu in the latest [Release v1.0.9.7_9c](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9c). Marlin Already has some stats and the only missing is the traveled meters by the axis but is what we have implemented. Let me know if we can close the issue.

[image]

Dan263737, 2025-03-08:
Thanks. works well.

closed 2025-03-08 by navaismo