navaismo/Ender-3V3-SE #49: Shortcut for other filaments preheat.

Dan263737, 2025-03-06:
Hi, i will like if you please add preheat shortcut for other filaments. There are now shortcuts for PLA and TPU - so it will be good if you add presets for PETG-240/90 and ABS-260/110. Thanks

eduard-sukharev, 2025-03-06:
There's already preheat setups in Marlin and its number is configurable and their params can be adjusted in M-code and store in eeprom. Generating a list of presets from what saved i eeprom can be done.

navaismo, 2025-03-08:
Added the corresponding menus in the latest [Release v1.0.9.7_9c](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9c). I got some issues with my printer and the EEPROM so please check if its saving values correctly and let me know if we can close the issue.

[image]

Eitschman, 2025-03-08:
Hi. The new menus work well, but there is now a rendering problem. I think this happens because the original menus for PLA and TPU (and others) in the German language are two lines long...

[image]

Dan263737, 2025-03-08:
Thanks .works great

navaismo, 2025-03-08:
> 
> Hi. The new menus work well, but there is now a rendering problem. I think this happens because the original menus for PLA and TPU (and others) in the German language are two lines long...
> 
> [image]

Oh, let me think in a way to erase the last row and then render again the option menú. That happens when scrolling up or down or both?

Eitschman, 2025-03-08:
Both directions

navaismo, 2025-03-09:
Should be Solved with the [Latest Release v1.0.9.7_9d](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9d)

closed 2025-03-09 by navaismo

Eitschman, 2025-03-09:
The rendering issue is not solved, but i can live with that. Don't waste your time on this!

navaismo, 2025-03-09:
I dont see anymore the ghosting on my menus could be the size of the rectangle that erase it, can you show me a screen  to see where is not cleaning it?

Eitschman, 2025-03-09:
Here you can see it.

[image]

navaismo, 2025-03-09:
By any chance can you play with the size of the rectangles and compile it?
The function starts at https://github.com/navaismo/Ender-3V3-SE/blob/43a8b93eebbe43e7abd504b352a1fbfe92fc3b89/Marlin/src/lcd/dwin/e3v2/dwin.cpp#L7729

And you may need to play with the highlighted lines below, in my setup it does clean the ghosting:

```diff
  {
          if (HMI_flag.language < Language_Max)
          {
+           DWIN_Draw_Rectangle(1, Color_Bg_Black, 200, MBASE(0) - 8, 240, MBASE(0 + 2) - 12); // Clear previos line
            Draw_Menu_Line(0, ICON_SetEndTemp);
            DWIN_ICON_Show(HMI_flag.language, LANGUAGE_Hotend, 42, MBASE(0) + JPN_OFFSET);
            Show_Temp_Default_Data(0, 1);
            Show_Temp_Default_Data(1, 2);
            Show_Temp_Default_Data(2, 3);
          }
        }
        else if (index_temp == 7)
        {
          if (HMI_flag.language < Language_Max)
          {
+           DWIN_Draw_Rectangle(1, Color_Bg_Black, 200, MBASE(0) - 8, 240, MBASE(0 + 2) - 12); // Clear previos line
            Draw_Menu_Line(0, ICON_SetBedTemp);
            DWIN_ICON_Show(HMI_flag.language, LANGUAGE_Bedend, 42, MBASE(0) + JPN_OFFSET);
            Show_Temp_Default_Data(0, 2);
            Show_Temp_Default_Data(1, 3);
          }
        }
        else if (index_temp == 8)
        {
          if (HMI_flag.language < Language_Max)
          {
+          DWIN_Draw_Rectangle(1, Color_Bg_Black, 200, MBASE(0) - 8, 240, MBASE(0 + 2) - 12); // Clear previos line
            Draw_Menu_Line(0, ICON_FanSpeed);
            DWIN_ICON_Show(HMI_flag.language, LANGUAGE_Fan, 42, MBASE(0) + JPN_OFFSET);
            Show_Temp_Default_Data(0, 3);
          }
      ```

Eitschman, 2025-03-09:
I will try it as soon as possible. This week is full of work, maybe in the evenings. I let you know.

eduard-sukharev, 2025-03-11:
I had similar problems during development. It was some or other code error that made the firmware billable, but not bootable. Check that you don't have any edits in the code. Check that you're on the right git branch. Can't recommend anything specific, I had mine the same way due to my own faults in code (don't remember what it was)