navaismo/Ender-3V3-SE #53: Localisation for new menus

Eitschman, 2025-03-08:
It would be very nice if all the wonderful new menus could be displayed localised. 
For the German language I could help.

Regards 
Andreas

eduard-sukharev, 2025-03-08:
The problem here is that custom menu items are rendered as text directly from MCU side, while original menu items are rendered as pre-rendered images with text, baked into the LCD itself and only rendered with commands like "draw this image" sent from MCU. Marlin, the original firmware for MCU, has predefined set of languages and translations for them, but one has to define them during the firmware build time in order to save memory. Switching between images is cheap - LCD firmware has separate image "banks", one per language, while items IDs are same between them. Switching languages means only selecting image bank 3 instead of 1, for example. Switching between different sets of strings means they have to be stored somewhere, and MCU SRAM is not the right place to be occupied by usuned sets of languages.
@navaismo can you think of any way to dynamically load language strings on startup? Maybe dumping all languages to SD card beforehand and read them on boot?

Eitschman, 2025-03-08:
Well, i think the language will not be switched at runtime or so. Can we just define translation-files that will be used while compiling the firmware? And then build separate files for the "main" languages? Just an idea...

navaismo, 2025-03-08:
> [@navaismo](https://github.com/navaismo) can you think of any way to dynamically load language strings on startup? Maybe dumping all languages to SD card beforehand and read them on boot?

We can do that, but the main problem will be to render big sentences like the German Menus that are 2 lines, which led us to another problem adjunsting dinamically the space between menus.

> Can we just define translation-files that will be used while compiling the firmware?

As @eduard-sukharev  explained is difficult to render more items without accessing the LCD RAM. 
Since dealing with the LCD is our main problem and people would like to have more features and of course localizated maybe will worth to call for UI designers and as Eduard and I [discussed here](https://github.com/navaismo/Ender-3V3-SE/discussions/28#discussioncomment-12348859) maybe worth to rewrite the TFT firmware with something more robust and with access to the LCD RAM, but this is not a 2 man job, we will need help of course.

I was just doing all this changes just to have something solid, but I see it growing faster and I dont have much free time to attend all. My next goal is to focus on a new Octoprint addon to integrate the ADXL G sensor and the printer to improve input shaping. So probably I will stop taking more request in the meantime depending on the complexity of it.

Eitschman, 2025-03-08:
It was just an idea. I personally do not have any problems with the English language and can live with that. I just thought of those people who have problems with foreign languages...

Close this request if it is too complex. 

I understand your problem with free time, I have the same. 
Maybe I can step into Marlin development myself. I'm a web developer with skills in Javascript and HTML/CSS, and a bit Delphi. So coding in general is not a problem.

navaismo, 2025-03-08:
Our main problem is this cheap closed chinese LCD which lack of documentation and all features are obscure, our life will be easiear with something functional like the LCDs that they use for mriscos firmware, you can play around and do all this sort of stuff. 

Guess we will have to hack it and build our own TFT file for it. I dont see another way around :(

eduard-sukharev, 2025-03-08:
We could setup a Github Actions to auto build firmware for each tagged release, but having `Number_Of_Languages x number_of_baudrates` will probably be an overkill and. bit overwhelming for newcomers.

Regarding the multiline we could implement the same thing they did in stock Marlin for 128x64 displays - just scroll the row upon highlight/selection. That would not be very simple, but doable. Loading different languages at runtime is a tad more tricky.

eduard-sukharev, 2025-03-08:
@navaismo you should probably define milestones for releases and prioritize tasks in backlog - that way you can clearly show that any feature requests are welcome and we can discuss them, but they will stay in backlog until some future release or anyone willing to get them sooner steps in to implement that.

navaismo, 2025-03-08:
> [@navaismo](https://github.com/navaismo) you should probably define milestones for releases and prioritize tasks in backlog - that way you can clearly show that any feature requests are welcome and [...]

Yep, I think the [Repo KanBan](https://github.com/users/navaismo/projects/1) is public(is it?) so people can see there what is in backlog, progress and Done.

closed 2025-06-27 by navaismo