navaismo/Ender-3V3-SE #56: Pause button is shown as Print until interacted with

eduard-sukharev, 2025-03-10:
## How to reproduce:
 
- Navigate SD card with files and select any file
- Confirm selected file to be printed

## Actual result

The print job screen shows 3 buttons:

- Tune (focused)
- **Print**
- Stop

If encoder is rotated CW the focus moves from Tune to the second button, which is now updated and becomes **Pause**

## Expected result

The print job screen shows 3 buttons: 
- Tune (focused)
- **Pause**
- Stop

navaismo, 2025-03-19:
This part of code render the buttons:

```c++
  // Setting interface
  ICON_Tune();
  if (printingIsPaused() && !HMI_flag.cloud_printing_flag)
    ICON_Continue();
  // pause
  if (printingIsPaused())
  {
    Show_JPN_pause_title(); // show title
    ICON_Continue();
  }
  else
  {
    // Printing
    Show_JPN_print_title();
    ICON_Pause();
  }
  // stop button
  ICON_Stop();
```

Basically the state of printingIsPaused is Unknown, and then when we move the encoder to it falls in the else to set the Icon Pause, made a quick test to try setting ICON_PAUSE by default but it get override by the mix of those IFs maybe a Switch is better or a first state var to render at the beggining the pause icon always or reverse the if.

AromaticArea3836, 2025-03-20:
@navaismo you can simply make an function, like:
`void PrintStart() //A function after gcode selection or print start
{
   ICON_Pause(); //Show "Pause" button instead of "Print" button
}
`
I am not so good at coding, it is only suggest, you can ignore this post if you want.

navaismo, 2025-05-02:
It is supposed to be fixed lets see from people that use SD if no bugs

closed 2025-05-02 by navaismo