navaismo/Ender-3V3-SE #57: Show bed adhesion help if print cancelled during the 1st layer

eduard-sukharev, 2025-03-10:
The most popular reason why the print was cancelled during the first layer is poor bed adhesion. It might be due to disabled bed leveling (after upgrade from 4x4 or 5x5 to 7x7 bed level grid), or due to not calibrated Z-Offset, or even plainly greasy bed surface.

We should add a wiki page for these cases and if print was cancelled early (first layer, based on parsed model metadata or just couple of minutes, excluding preheat time) then show helpful message with link to the wiki page.

This should help inexperienced users with problems they might be encountering.

Eitschman, 2025-03-10:
I think this is a good idea. But when you are going to implement that there should be a option to disable the message. I think an experienced user would be nerved somehow, when there is always a message popping up.

navaismo, 2025-03-10:
> I think this is a good idea. But when you are going to implement that there should be a option to disable the message. I think an experienced user would be nerved somehow, when there is always a message popping up.

Like me when I saw that weird Uknown ERROR message popup and my printer got useless until i powercycle it xD

eduard-sukharev, 2025-03-10:
Yeah, that unknown error is my fault, I really should fix it ASAP. I got the same one with thermal runaway - my heater and thermistor were disconnected and the bed leveling procedure starts heating.

navaismo, 2025-03-10:
Speaking of bed leveling, What about a menu to render the Mesh like Octoprint Bed level visualizer does?
Should be possible right rendering line by line as the thumbnail what do you think?

eduard-sukharev, 2025-03-10:
You mean like 3D height map? Kinda doable, but I guess of dubious value: the map will only show that it's not completely plain and actually editing those height values is going to be a mess to visualize. Or just a menu option to render it for fun? well, that might be ok, but that would require some knowledge of 3D rendering at least - setup camera position, angle and aperture, render an exagerrated height map onto a camera view and then send it line by line to LCD. I wouldn't use that and I wouldn't want to code that, but that's somewhat a CGI intern would be successful at :)

navaismo, 2025-03-10:
> Or just a menu option to render it for fun?

Yes, for fun but to show how crooked is to make an idea of what to adjust. Nothing fancy but something useful. 
could be a nice addon to waste 2 mins seeing how it renders.

Wingtiger, 2025-03-11:
I don't like this idea.

navaismo, 2025-03-11:
> I don't like this idea.

That's why when we implement it we will create as @AromaticArea3836 suggested a menú to disable Help Popups.

eduard-sukharev, 2025-03-11:
> I don't like this idea.

Can you be more specific about what exactly you don't like about it?
The QR codes in UI? The lock to this repo wiki? The constant popup if cancelled early?

Wingtiger, 2025-03-14:
There is no need to clutter the microcontroller's memory.

Dan263737, 2025-03-31:
Sound like a cool idea for beginners! It will be more user friendly too.

closed 2025-07-01 by navaismo

eduard-sukharev, 2025-07-01:
@navaismo the latest PR was merged and we're closer to better newcomers experience. Should we make the "Advanced help messages" default value to true so that it can be disabled by those with experience and still provide intended value by default to the new users?