navaismo/Ender-3V3-SE #58: Setup github actions to build firmware with 3 baudrates on every release

eduard-sukharev, 2025-03-10:
Currently all the prebuilt firmware binaries are built manually, uploaded as files and stored in Firmware folder.

It's not the best way to distribute firmware. We could leverage the power of github actions to build firmware and just upload artefacts on releases. That would remove manual work done upon each release.

We could reuse existing platformio actions to build:
```yaml
jobs:
  build:
    runs-on: ubuntu-latest
    steps:
      - name: Checkout
        uses: actions/checkout@v3
      - name: PlatformIO Run
        uses: karniv00l/platformio-run-action@v1
        with:
          silent: true
          verbose: true
          disable-auto-clean: false
```

we would need to figure out how to handle the different baudrate settings for matrix build and upload artifacts.

navaismo, 2025-03-11:
Hi, I'm not planning to use github actions for time being. I'm comfortable shipping binaries as the way it has been and also people can make the fork/download the code and build themselves.

closed 2025-03-11 by eduard-sukharev

eduard-sukharev, 2025-03-11:
I'm ok with the naming - as long as people keep using it and we can refer to it by some already popular name it's all good.
I am on the kanban board, but it somehow doesn't allow me adding items to it - I can create and assign milestones, but cannot put items to backlog.

I would definitely advise automating as much as possible - building manually is error prone, we're all humans after all. And doing releases using the auto built artifacts is a more streamlined and commonly used way of maintaining big projects. But it's up to you to decide.