navaismo/Ender-3V3-SE #61: Thumbnails not work

sy-per, 2025-03-11:
Hello, I have tried several baud rates, but none of them display the thumbnails, and octoprint blocks printing until I exit the rendering.

[image]
[image]
[image]

navaismo, 2025-03-11:
Hello!

Are you using the firmware version that enables the Thumbnail on printer side?

This one: https://github.com/navaismo/Ender-3V3-SE-Gcode-Preview

sy-per, 2025-03-11:
Ohhh no I use the version from here:
v1.0.9.7_9c
https://github.com/navaismo/Ender-3V3-SE

It doesn't work with the thumbnail?

navaismo, 2025-03-11:
Nope, you need the other firmware version to render it.

closed 2025-03-11 by sy-per