navaismo/Ender-3V3-SE #65: My printer *sounds* different now after switching from factory stock to this + octoprint

ronniedude, 2025-03-14:
I know this sounds stupid but after updating from Factory 1.0.4 -> then Creality 1.0.6 -> finally navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9d + Octoprint.

I know this adds a lot of variability to the cause, could be the speed limits I set in Octoprint (left them as default)

My Stepper motors sound different. If anything they sound worse, more noisy & almost *crunchy*(?).

Anyone have any thoughts or ideas to this?

navaismo, 2025-03-14:
We have increased the travel speed for the Homing Macro. But the printing speed is set by your slicer.

> could be the speed limits I set in Octoprint (left them as default)
Can you elaborate about that?

>My Stepper motors sound different. If anything they sound worse, more noisy & almost crunchy(?).
If the belts are thight that could cause the sound.

ronniedude, 2025-03-14:
> > could be the speed limits I set in Octoprint (left them as default)
> 
> Can you elaborate about that?

In the first-time setup wizard for Octoprint the defaults are 6000 mm/min

> > My Stepper motors sound different. If anything they sound worse, more noisy & almost crunchy(?).
> 
> If the belts are thight that could cause the sound.

I don't this this is it because the sound changed only after software upgrade from stock 1.0.4 -> everything after

navaismo, 2025-03-14:
> In the first-time setup wizard for Octoprint the defaults are 6000 mm/min

No issues, settings are from slicer.

>I don't this this is it because the sound changed only after software upgrade

Thats my point, before upgrading you had dialed the belts for the old speed, now you have another speed set, you need to double check the tension on your belts and also the rod grease, as explained above now the printer has increased the speed for homing.

eduard-sukharev, 2025-03-14:
You could also check maximum acceleration - this fork has max acceleration increased which means whatever defaults were in your slicer could've been capped by stock limits but now increased sure to raised limits. If you wish you may just set max acceleration in Settings menu to the stock defaults.

AdamDove, 2025-03-16:
I've noticed the same noise change between this version and a version I was using from January.  I'm going to dive into the max acceleration settings I use in Cura and report back tonight.

AdamDove, 2025-03-17:
I've done a diff between 1.0.9.7_4b and the latest to see if I could find any code differences that may account for the increased noise, and I've come up empty, so I did some more digging into eduard's comment about max acceleration.  After a few tests, I determined that Cura doesn't set acceleration for either print or travel by default, and instead relies on the firmware acceleration setting.  Changing the max acceleration on the printer to 2500 mm/s/s for both x and y (the original firmware default) returns the printer to it's former silent movement.

I have no clue why I thought it was quieter on 1.0.9.7_4b.  That part must have been in my head.

For me, I'm now going to go back to 4000 or near to it as I had Linear Advance tuned nicely for it prior to these tests.

I'd recommend closing this issue, however adding the following comment to the README.md would be helpful to future users:
"This firmware increase the default max acceleration from 2500 mm/s/s to 4000 mm/s/s.  This setting is changeable from the Motion menu on the printer.  Some users report increased noise at this higher acceleration."

closed 2025-03-19 by navaismo

Dan263737, 2025-05-03:
Just have the same think when updating from V1.0.9.7d from V1.0.9.8

navaismo, 2025-06-10:
The [Latest Release v1.0.9.8_1c](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.8_1c) now split the Firmware in two:

- Normal: All feedrates are the same as in stock firmware.
- Fast: All feedrates has been increased for those who wants speed.

> [!IMPORTANT]
>
> - [x]  When choosing betwen Fast or Normal FeedRates, after flashing the new firmware **YOU MUST** reset the printer so the default values are loaded again.
> 
> - [x] All issues opened without reading release notes are going to be closed automatically.

navaismo, 2025-07-20:
> its because the stock firmware ONLY uses Stealtchop(driver for stepper motors), while the custom firmware when printing over 100mms it changes to spreadcycle and spreadcycle is WAY more loud than Stealtchop.

Yes, its explained here: #138 

> I have put the Stealtchop limit at 300 mms,(only at my bed i couldnt hear anything wierd from the other axis)
> 
> Here is the gcode command's:
> 
> M569 P1 V300 ; keep stealthChop up to 300 mm/s M500 ; Save to EEPROM

Be careful with the values, while Stealthchop is quieter it is meant for lower speeds. Increasing the threshold by 3 times could affect the print quality since forcing speeds >150mm/s could cause missing steps and loss of the torque; which could cause layer shifts or some other undesired artifacts on your object.