navaismo/Ender-3V3-SE #67: Cura Slicer cannot use the USB function for sending print jobs

Shushkata, 2025-03-16:
After some fiddling i managed to install the firmware. I am really exited for all the new options. As i am still researching what and how i noticed that the printer is no longer visible in Cura Slicer when connected to the USB C port of the unit.  Do i need some drivers, or this is something that got sacked on the way :) ?

navaismo, 2025-03-16:
Depending on how Cura connects you should check the baudrate that you flashed on the printer and double check in cura that is the same.

Shushkata, 2025-03-16:
I flashed the recommended 150000, so i found the file in Cura responsible for the baudrate and in the list there is no 150000. Adding it does not do the trick. Should i reinstall with a different baudrate ?

closed 2025-03-16 by Shushkata