navaismo/Ender-3V3-SE #71: some Ideas to add and some question

adhamwira, 2025-03-19:
```
Hello this is my first time posting on github. sorry for bad english
I have a request to add some features and changes that may improve this firmware also some question
```
### My Ideas
- **Faster Homing** I feel like the homing process XYZ can be sped up a little bit more for faster homing specially the Z axis. Is it possible?

- **Different Position For Custom Extrusion** I have a 3d printed side bin installed from printables and I tried to extrude the filament above the bin at Z20 X-12 Y-12 and I think its better than having to extrude above the bed position so I don't have to touch the bed to clean.

- **Nozzle Cleaning Option Before Print** I used to leave my printer after pressing start print and make sure the nozzle clean but there is still a problem where the nozzle squirts some filament when heating the nozzle and ruin my first layer. I have an idea to add an option to auto clean the nozzle before first line on the left by using the cleaning mechanism from the build in auto homing process. I think it will make the first layer safer and cleaner. just an idea hehe

### Question

- **Stepper Motor Enggages when changing the Z offset** when I change the Z offset on the main menu (not printing) the motor engages automatically is that normal?
- **Input Shaper Default Setting?** is the build in input shaping value or setting is correct or do I have to set to 0 because I don't have accelerometer to calibrate it? thanks

eduard-sukharev, 2025-03-19:
it's better to split this request into three different request - one per your new idea so that they could be discussed separately and their implementation could be tracked individually

eduard-sukharev, 2025-03-19:
Changing Z offset effectively raises and lowers the nozzle with regard to probed level - it's ok that it engages Z motor. Use disable stepper in menu to disengage it.

eduard-sukharev, 2025-03-19:
Input shaping is pretty much close to what's set in upstream Marlin firmware, which is supposed to be an average for some generic printer. You may set it to 0 to disable it or calibrate it using calibration procedures even without accelerometer. Adding to or removing things from the stock print head will ultimately change the calibrated value so one has to recalibrate it again afterwards.

eduard-sukharev, 2025-03-19:
Regarding the speed limits — no movement is instantaneous and involves acceleration.
You can read more about it on [our wiki](https://github.com/navaismo/Ender-3V3-SE/wiki/Printer-acceleration) now.

AromaticArea3836, 2025-03-20:
I have one request: Add a homing speed(acceleration) setting with a name, like: "Homing speed" or "Homing acceleration" in _PREPARE_ or _CONTROL_ menu, that will be limited to: 
`X axis: maximum 2000`
`Y axis maximum 2000`
`Z axis maximum 350`
This menu will look like a acceleration setting in `CONTROL` menu.

AromaticArea3836, 2025-03-20:
+ By a auto-leveling process, those settings will be ignored(default settings will be used), and after auto leveling, those settings will be on!

navaismo, 2025-03-21:
> Add a homing speed(acceleration) setting with a name, like: "Homing speed"

Unfortunately Homing doesnt work like that in the code; but the global acceleration it does affect it. 
Marlin set the "_speed_" of homing in terms of mm/s right now is if I'm not mistaken the homing speed is set to 50,50,8 mm/s for X,Y and Z axis respectively.

And the G28, the homing feature has some controls to make slow / smooth motions to avoid issues and to be able to detect on time the limit switches. So this is something that I will not recommend to left to user to set. 

We may increase the speed to 80,80,80 to see if that is better but overall we will need a community feedback on this.

artsitemail, 2025-03-21:
I don't know if the ender 3 V3 SE uses Jerk or junction Deviation, but would it be possible to put a menu on the printer so that it can be modified manually without using the slicer?

I'm having problems with a piece that has a 60º angle with overhangs and that could be the problem.

By the way, thank you for the excellent work done with the firmware, something that should have been done by Creality.

artsitemail, 2025-03-21:
No, a little in the walls but I turned on extrusion rate smoothing at 8 and increased the jerk to 10 and it improved a lot.

AromaticArea3836, 2025-03-31:
I like the idea about nozzle cleaning. We can do this before reading the gcode. Simply do like bambu lab A1. Heat up the nozzle to 240(not 250 because of PTFE), bed 60 and do a clean with fast movements at the back of the build plate(again, like bambu labs).

navaismo, 2025-05-02:
> * **Faster Homing** I feel like the homing process XYZ can be sped up a little bit more for faster homing specially the Z axis. Is it possible?

Implemented in the latest [release 1.0.9.8](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.9.0.8).

But you will need to adjust the `Max Speed` under `Motion Menu` and in your slicer if it propagates the limits. For example to make Z faster adjust the motion to 15 and you will see faster moves. Ensure your slicer dont overwrite the settings. Orca does and you need to disable or set the speed there too under machine settings.

> * **Different Position For Custom Extrusion** I have a 3d printed side bin installed from printables and I tried to extrude the filament above the bin at Z20 X-12 Y-12 and I think its better than having to extrude above the bed position so I don't have to touch the bed to clean.

Now is going to the left before start extruding.

closed 2025-05-02 by navaismo