navaismo/Ender-3V3-SE #72: End Print menu is too early

AromaticArea3836, 2025-03-21:
Hi! Menu "Print successful" or "Print complete" is showing too early. in stock firmware this menu shows only after print present by a printer, when printer's bed moves forward at maximum range(220?). A video example will be a little later.

navaismo, 2025-03-21:
Using Octoprint Plugin will appear right after the head finishes the last layer, which is when the printer notify the Host about it. Speaking in terms of visible layers since we have a small delay updating the current layer is 99% possible that you will see it finishes before reaching the N-last layer but it is only due to that delay.

AromaticArea3836, 2025-03-21:
@navaismo then we can close it, right? It is very funny to have a small issue here :)