navaismo/Ender-3V3-SE #73: Different speeds when stoping the print

AromaticArea3836, 2025-03-21:
Hi again! Remembered about one bug(? If not, you can close this issue as well), that if I stop one print, it will present the print(move print bed to 220) and it will happen with one speed, for example, 150mm/s or 1500mm/s/s. But when I start another gcode file, I will get other speed of presenting the print for example, 75mm/s or 750mm/s/s. Those speeds are slower or faster. It can happen on the same gcode, but in different height, for example.

navaismo, 2025-03-21:
Are you using OctoPrint?
If you used the button 'Cancel', then the Cancel macro uses the one defined in the GCODE SCRIPTS section and depends how it is configured. 

If you used the cancel button on the LCD it will use the stock firmware Macro to move the bed and head which indeed is slooow and only notify the Host about it.

AromaticArea3836, 2025-03-21:
No octoprint. Clicking stop button on LCD.

AromaticArea3836, 2025-03-21:
Is there any way to change this macros to get the same speed?

navaismo, 2025-03-21:
If using the normal menu from LCD is the same Macro, should be the same speed everytime.

navaismo, 2025-03-31:
@AromaticArea3836 do you have some steps to reproduce the issue?

I've tried two prints and both resulted in the same speed/movements. How did you noticed that.

eduard-sukharev, 2025-03-31:
> @navaismo Oh, I forgot to read github at all. I will do a video of this problem next day.

A zip with files to print and thorough steps to reproduce would be much more appreciated!

AromaticArea3836, 2025-04-01:
So I understood this problem. Print present speed after stopping is set by last speed change of gcode at the moment. So this macro that is stopping the print dont set the speed of presenting the print

closed 2025-04-01 by AromaticArea3836