navaismo/Ender-3V3-SE #74: Stored features does not load consistently.

Shushkata, 2025-03-21:
Hi, when i set something - speed, acceleration, etc. on the printer and save the config, if i restart the printer most of the time the printer settings are reset to the defaults of the firmware. From time to time i also see that the stored setting are also lost and i need to dial them again (somehow loading settings result in loading defaults again). Other settings, like the Linear Advance are kept intact despite turning on and off the printer. 

Am i doing something wrong or this can be an issue somewhere in the code ?

navaismo, 2025-03-23:
We just added some extra values to store, the rest are the default routines. The only way to check if your printer is saving the values is connecting a serial terminal.

Edit a value and store the configuration, then read the configuration, after that all saved values will be shown in the serial terminal, check if the values match. Finally turn off the printer, turn on again and validate if the values are there if not I will say is somthing with the eepromm.

Shushkata, 2025-03-23:
Unfortunately i am not that aware what to do. I just powered the printer to see that acceleration and speed are again reset to their defaults. Maybe if you can point me in the right direction - if the serial terminal is software based i will eventually figure it out if i need an external device i don't own anything special.

navaismo, 2025-03-24:
Upon further investigation, I can confirm the issue. Values are stored but when the printer is powercycle and the settings are loaded again it overrides everything with the defaults.

I'll check in which routine do that and why?

navaismo, 2025-03-24:
Ok Found the "issue".
At least in my case when I send a file to print the GCODE file changes back my settings:

```python
;EXECUTABLE_BLOCK_START
M73 P0 R115
M201 X2500 Y2500 Z500 E5000
M203 X250 Y250 Z5 E40
M204 P2500 R500 T2500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
```

Seems like orca add this by default and then the printer override the previous values I noticed because I changed the Z speed and suddenly after printing something it went back to the default.

Shushkata, 2025-03-24:
So this is on the slicer end, not the eeprom itself ? I was perplexed as i use Cura and Orca. That is way it was from time to time and not always when i restart the machine. 

BTW i found something, than maybe is by design. Cura communicates with the printer with the USB C interface. When it does on the screen (of the printer) i get - Octo Print Connected, where i don't have this installed in any way. As long as the USB C is active, i cannot load and start any print jobs from the SD / TF card. It does not do anything. So if i want to run Orca i need to unplug and restart so it can accept the card. 

This is not related to the problem, but did not want to open a new thread.

navaismo, 2025-03-25:
> So this is on the slicer end, not the eeprom itself ?

Correct, you can double check it though.