navaismo/Ender-3V3-SE #78: Silent mode

Dan263737, 2025-03-30:
Hi, can we make something like a silent mode? I think sonetimes the noise from printer is really annoying. It may contain: lower axis speeds, slower speed of hotend fan or manually control hotend fan and lower the jerk so it will move more smoothly or stealthcrop motors setting. Thanks

eduard-sukharev, 2025-03-30:
Silent mode is just a preset of speeds and acceleration in your slicer. Limiting this in firmware doesn't make much sense.

Dan263737, 2025-03-31:
Prusa mk4 has something like this.

I mean silent mode for homing, leveling and things like this not for print.

[image]

eduard-sukharev, 2025-03-31:
Ok, this might need more research. No promises here.

Dan263737, 2025-03-31:
> > Ok, this might need more research. No promises here.
> 
> I'm still thinking should be not related to the firmware but user/slicer settings.

Yes, but i think silent mode for the non-print movements. (My is often very noisy).

navaismo, 2025-06-10:
Due to memory issues This feature cannot be added. The suggestion to make it less noisy is to use the Firmware from "Normal Folder" and then adjust lower all the acceleration and speeds to make it less louder.

closed 2025-06-10 by navaismo