navaismo/Ender-3V3-SE #79: Question about input shaping calibration.

Dh0mp5eur, 2025-03-31:
Hi,

I have carried out calibration tests for the first time and obtained the following values (X 54.47 / Y 24.17).

[image]

[image]

But the calibration greatly degraded the quality of the surfaces.

[image]

I reset the settings to their default values to get a clean result.

I've read that it's preferable to recalibrate the linear advance after input shaping calibration.
But seeing what I'm getting, I think there's another problem.

Is my calibration wrong or is there something wrong with the input shaping?

I've seen that Marlin doesn't yet support calibration with an ADXL (work in progress with Octoprint).
Is there a better calibration technique?

Thanks in advance

navaismo, 2025-03-31:
Yes, you need to recalibrate Linear Advance too,  with the input shaping disabled (M593  F0) after you found the values.
Using the ringing tower is kind of tricky too and to have better results is recommended that  you stiff the gantry of the machine, there are some  mods that doesnt require drilling https://www.printables.com/search/models?ctx=models&q=gantry+v3+se 

Finally if I recall correctly you need to set not only the Frequency of axis but the Damping ratio too, could come from there the issue as well.

Dh0mp5eur, 2025-04-01:
Thank you navaismo,

I have already reinforced my printer with a gantry and replaced the X/Y axes with linear rails.

I changed my calibration method for this one, it's easier to determine the right frequency and damping.
https://marlinfw.org/tools/input_shaping/freq-calibr.html

For the moment I still have a rougher surface on the top layer than I had before.

Je continue à optimiser.
I feel like I need to increase the flow.

Dh0mp5eur, 2025-04-01:
After too many tests, I went back to the default setting.

I couldn't find a setting that would give me a smooth top surface.

Thanks anyway.

navaismo, 2025-04-22:
Hi, if you have a raspberry pi 3B or superior, the accelerometer and willing to test here is the Release Candidate for the Input Shaping plugin for Octoprint.

https://github.com/navaismo/Octoprint-Pinput_Shaping

closed 2025-04-28 by eduard-sukharev