navaismo/Ender-3V3-SE #80: Idea / Make default Z height after auto home and extrude commands greater

Shushkata, 2025-03-31:
I have a proposal. After you initiate a home command , the head finishes in the middle of the X/Y axis with a Z value of 10. This is compact and looks good, but it's a problem. If the printer is heating it's nozzle,  filament start to come out, and there is just not enough space to pull it out before the printing process starts. So it starts to drag some excess filament to the stating point.  ( i know this can be changed with the G Code, but bear with me for a bit more) 

Same and also really annoying is the extrude command, where the head is again going in auto home position with the same Z=10 and starts to extrude filament. It is problematic as the space is not enough for the filament to expand and from time to time it can stick to the nozzle and secondly it's really hard to clean it (as after the extrusion the nozzle temp goes down to 140 and the filament stops melting, so you need to do it right away)

My idea is , to change the default Z=10 so something more reasonable - 30 maybe, so you can put at least two fingers under the head. This way you don't need to constantly move the head manually up so you have more clearance between the bed and the nozzle.

navaismo, 2025-03-31:
In that case you can use the custom extrude option it raise the head to 35mm

Shushkata, 2025-03-31:
Yes i forgot about it. This will eliminate one of the issues u face, but maybe for the default state it will be beneficial to have it higher.

navaismo, 2025-03-31:
Ok, I will increase the Z height on Homing on the next release.

navaismo, 2025-05-02:
Implemented in the latest r[elease 1.0.9.8](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.9.0.8). This feature is user editable so everyone can decide the height.

[image]

closed 2025-05-02 by navaismo