navaismo/Ender-3V3-SE #86: Remaining time to print always off

RamenNibbler, 2025-04-12:
At the end of every print the remaining time of printing is always off by 10% of the elapsed time, my printing progress reaches 99% but remaining time is still 20mins for example.
I recently encountered a bug where I printed a small figure (< 0.5 g), the remaining time showed " >100H "

Excuse my bad English :)

xnggypr, 2025-04-14:
Why do you close this issue?

xnggypr, 2025-04-14:
I have the problem, when remaining print time is under 1-2 minutes and remaining print time became 100H>

RamenNibbler, 2025-04-14:
I thought it was a one time thing or a something wrong while slicing

navaismo, 2025-04-15:
@eduard-sukharev is investigating this on the issue on his free time to help him please share the Gcode file that has the issue and the Slicer from it.

this month has been hectic for both of us so expect delays on fixes and releases.

eduard-sukharev, 2025-04-15:
The reason for ">100" in really small models is that they were sliced in slicers that don't put model meta info at the start of gcode file (Orca Slicer is the one), and thus firmware can only guess. Guestimate is based on the file size, time already spent and the remaining file size. To avoid division by zero we only show time remaining after 0.1% (or 0.01%, can't remember from the top of my head) is done, until then it's unknown and even after that it's just a guess and having a good example gcode file to reproduce on our side would greatly help.

RamenNibbler, 2025-04-15:
Forgive me but I'm new to this, when you say share the g-code, is it the 3d models gcode or what are you referring to

eduard-sukharev, 2025-04-18:
The gcode is a file ending with `.gcode` you get from slicer, put on the SD card and start printing. If you print from slicer directly, then use Export G-Code or Save to file feature. Put that file into a ZIP archive and attach it here.

RamenNibbler, 2025-04-18:
[E3V3SE-Spool_Holder (1).zip](https://github.com/user-attachments/files/19812937/E3V3SE-Spool_Holder.1.zip)

RamenNibbler, 2025-04-18:
[image]

xnggypr, 2025-04-18:
Ahh, yeah forgot about 2rd variant. In the last layers, LCD shows the print_time(from slicer) - current_time, like slicer said, that this model will be printed 3h 30m and current time left is 2h50m, but it's last layers. Then the LCD will show  3h30m-2h50m=0h40m left.

eduard-sukharev, 2025-04-18:
Ok, that's something to work with, thanks for your input.

The issue here might probably be because, to eliminate the difference in printing time or something, the actual print progress does not take into account the heat up time.
Also, there might be an issue with the slicer estimation - it might just not know exact capabilities of your printer. Did you enable the acceleration control in slicer? Are your acceleration and linear / pressure advance settings applied during the print?

RamenNibbler, 2025-04-18:
I mainly use PrusaSlicer with a profile for Ender 3 V3 SE, Cura as well, I just change basic settings like infill, shells and cooling that's it.

RamenNibbler, 2025-04-19:
An update.
Tried out OrcaSlicer this time got pretty good prints and most importantly accurate remaining time for some reason.

eduard-sukharev, 2025-04-20:
> I mainly use PrusaSlicer with a profile for Ender 3 V3 SE, Cura as well, I just change basic settings like infill, shells and cooling that's it.

> An update. Tried out OrcaSlicer this time got pretty good prints and most importantly accurate remaining time for some reason.

Did you notice if printing with one or another was quieter or louder? My guess is that PrusaSlicer assumes and doesn't enforce the accelerations settings and uses stock defaults (which are lower and more conservative speed-wise) and thus expects longer print times, while OrcaSlicer has acceleration settings enabled and thus whatever they are set in printer, are overriden by slicer in gcode and thus are closer to what slicer expects.

Can you verify the acceleration control is enabled or disabled in either slicer? Can you check print times when it's enabled/disabled?

closed 2025-12-26 by navaismo