navaismo/Ender-3V3-SE #88: Revisiting "Double-take on pause" (#32)

andersborgabiro, 2025-04-15:
This video shows how it behaves when performing a filament change. I forgot to remove the extruded filament, but other than that the prints succeeded.

https://imgur.com/a/ender-3-v3-se-post-filament-change-dKyPC0f

My G-code Scripts in OctoPrint only contain M117 and M300.

In the slicer (Orca) there's just an M600 in the Change Filament G-code, and there's an M25 for Pause.

You mentioned "out of filament behavior" earlier. I don't have a filament sensor, but doesn't it have to be activated via G-code anyway?

andersborgabiro, 2025-04-15:
Thanks for confirming. I'm doing a complete print job log now to see if anything odd pops up.

andersborgabiro, 2025-04-15:
The print job I filmed was a special one as I use Pause (M25) so I can insert an NFC tag into the print and later a Filament Change (M600) to change the color for the logo on top of the print.

serial.log only contains one M25 and one M600.

If I print something that only uses Filament Change this doesn't happen, so in the previous print job it seems the printer remembered that M25 was performed earlier and erronously performed it once again for Filament Change.

What's weird is that if I perform a new print job the M25 is no longer remembered, but maybe that's due to print job initialization.

andersborgabiro, 2025-04-23:
I've been able to verify this during several prints, with or without a Pause before Filament Change, so it's not a fluke. It's neither a major issue.

navaismo, 2025-04-23:
Can you share a GCODE file so I can check it?

andersborgabiro, 2025-04-23:
I used a model that I've seen shows this issue. Rather quick to print.

[serial - Print with Pause and Filament Change - 2025-04-23 - Anders Borg 2.zip](https://github.com/user-attachments/files/19868752/serial.-.Print.with.Pause.and.Filament.Change.-.2025-04-23.-.Anders.Borg.2.zip)

navaismo, 2025-04-23:
Any other model that you can share maybe something basic from printables that you add the command slice it and share, I want to print it direct to see it. Because everytime I slice & print a simple cube or any other gcode with M600, my printer doesnt do that, just behave like in the video I've shared.

andersborgabiro, 2025-04-23:
This is my reference cube: https://www.thingiverse.com/thing:6516157

[Reference Cube.zip](https://github.com/user-attachments/files/19872192/Reference.Cube.zip)

I used Orca for the 3mf and gcode.

andersborgabiro, 2025-04-23:
I didn't find anything obvious when walking through the code for M25 and M600 but I suspect a global variable that's not reset until a new print job.

navaismo, 2025-04-26:
@andersborgabiro Quick question: Are you using Ocotoprint or just the standalone ?

andersborgabiro, 2025-04-26:
Always OctoPrint

navaismo, 2025-04-26:
I have spent the whole morning trying to get the root cause of the issue, somehow `M24` mess with the `PARK_HEAD` position after we invoke `M25` command. So the next `M600` when resuming first goes to the default X0, Y0 position making the double movement.

So far I couldn't find how to override that behaviour because Printer/Octoprint thinks is an SD Card(standalone) command. But I have in the meantime a work around:

If you change the pause command from `M25` to `M0`  Octoprint will rise a warning message telling It is not going to send `M0` because is a halt command, but will execute the `PAUSE SCRIPT` defined in octoprint. Then you resume normal and the next `M600` will work as usual no double take/move on the head because M24 wasn't there to override the position.

Give it a try in the meantime I find where and why is overrriding the park after `M25/M24`

andersborgabiro, 2025-04-27:
Great. I'll try that.

navaismo, 2025-06-12:
I have changed the routines of PAUSE while using Octoprint to match the PAUSE SCRIPT set in the Script section of octoprint.
For that you will need to install

1. FW [Latest Release v1.0.9.8_1d](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.8_1d)
2. Plugin [Latest Release v0.2.7](https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails/releases/tag/v0.2.7)

> [!IMPORTANT]
>
> - [x]  Go to plugin settings and configure wheter to override or not the M25 on a GCODE FIle.
>

See if works so we can close this.

andersborgabiro, 2025-06-13:
Is Fast vs Normal build explained anywhere?

navaismo, 2025-06-13:
> Is Fast vs Normal build explained anywhere?

Yes, in the release notes: https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.8_1c

> ### Splitted firmwares between "FAST" and "Normal" feedrates. The Normal feedrates are the same as in the stock, so people stop opening issues that arent issues.

andersborgabiro, 2025-06-15:
Printer:
- Installed 1d (normal).
- Performed leveling etc before I tried to print anything.

OctoPrint:
- Installed the plugin.
- Tested with M25 disabled and enabled.
- Added custom Gcode for Pause/Resume etc as listed in the README.

The M25/M600 issue still occurs, whether M25 is enabled or disabled in the plugin, so I don't know what can be done for it to behave differently. As earlier, my guess is that it's due to global state inherited by M600, so an external fix shouldn't affect that.

Noted issues with the plugin activated:
- The OctoPrint file list sometimes doesn’t update for new jobs while the plugin is activated.
- There’s a warning (plugin conflict? not sure which one) that goes away if I deactivate the plugin: _Warning: A layer analysis error has been detected. You may have started the print before the gcode analysis had been completed, or the analysis may have failed for some other reason. (You can disable this warning in settings)_
- Sometimes I get a modal dialog over OctoPrint, so I hardly see the printer state and can’t control it. There were also round popups on the printer screen when Layer Progress was chosen, but not for Time Progress. I'm not sure about the distinction.
- In the settings dialog for the plugin the tooltips are squished to the right side so hard to read.

For now I will not use the plugin, and you can close this issue. I'll work around it, and I won't bring the issue up again.

closed 2025-06-15 by navaismo