navaismo/Ender-3V3-SE #96: The firmware cannot be replaced.

demxp, 2025-04-21:
I purchased this printer a few days ago, and was about to update the firmware, but I came across an unknown version V1.0.9_F401 and a recent revision of the motherboard. 

[image]

Apparently, it is built on the STM32F401. The version of this firmware is quite recent, it appeared on the website only on April 15th. Accordingly, the printer did not see the new version downloaded here. Is there any solution to this problem?

My version: SE_HWCR4NS200320C13C14_SWV1.0.9_GD303STM32F401 - 15.04.2025 - from: [https://www.crealitycloud.com/downloads/firmware/ender-series](https://www.crealitycloud.com/downloads/firmware/ender-series)

[image]

Hardware: cr4ns200320c14

Thanks a lot in advance!
Sincerely, Vladimir.

eduard-sukharev, 2025-04-21:
this firmware recently popped on reddit and it was said to be released for some new hardware invariant of Ender3V3SE. Sadly, no updates to original firmware source released by Creality was found and no further sources released, which leaves us wondering what that firmware is for and whether it's updatable.

This fork is for printers that came with original stock firmware.

eduard-sukharev, 2025-04-21:
After quick search up the mainline upstream Marlin there seem to be no pin definitions for this kind of board. Until then, we cannot build a version of this form for this hardware invariant.

Once such pins definition found, this could probably be done shortly.

demxp, 2025-04-22:
Thank you very much for your answer, I also wondered about this. When I was trying to figure it out, I downloaded this firmware, there was a README file in Chinese, here is the machine translation into English:

"The method of updating the motherboard firmware is as follows:
1. Put the unzipped folder STM32F4_UPDATE and Ender-3 V3 SE_HWCR4NS200320C13_SWV1.0.6_GD303_20230919.bin into the root directory of the storage, insert the TF / SD card interface, and the machine can automatically recognize and update;
2. Insert the TF / SD card, restart the machine and wait for the update to complete."

The old firmware, version 1.0.6 (but! with the date 20230919, while the site indicates the release date of 2023-10-19 - a month difference) and the update for STM are copied to the SD card. It's unlikely that the processors can read patches and apply them in their own flash memory, which means that it might first install "SE_HWCR4NS200320C13_SWV1.0.6_GD303_20230919.bin", which doesn't hurt work, and then update to "Ender-3V3 SE_HW_CR4NS200141C14_SW_V1.0.9_F401". Maybe that helps somehow.

eduard-sukharev, 2025-04-22:
There seem to be a lot of different boards by GigaDevice (as pointed out here in [preliminary support for v3 se](https://github.com/MarlinFirmware/Marlin/pull/27003) in upstream Marlin) and it'd be very useful for further investigation if you post a clear image of what you mainboard reads, especially the main chip markings.

demxp, 2025-04-22:
Took the clearest possible photo of my motherboard.

[image]

eduard-sukharev, 2025-04-22:
From what I've seen on the [upstream Marlin](https://github.com/MarlinFirmware/Marlin/blob/db3c5dd877fb41e2cd03ca3ff02c50e4f2bfcbbd/Marlin/src/pins/stm32f4/pins_CREALITY_F401.h#L26-L27), this seem to be the STM32F401RET6 from Ender-5 S1. Not sure how different is that from our older boards, but it seems doable. Backporting the board pins definition file should not be hard, we can look into that.

demxp, 2025-04-22:
Thank you so much for your attention to my problem! I'll be waiting for some news - your development is much superior to the original.

weirdal3333, 2025-05-02:
Creality removed the 1.0.9 firmware from their website, and archive.org does not have a backup of it... 
I like the idea of using a custom firmware, since releases can't be erased like it has. Do you think this'll be finished soon? I have heard amazing things about your firmware! 
I am hoping that octoprint will eventually be able to use your firmware as a base to make this machine more functional 😄

eduard-sukharev, 2025-05-02:
I've started the work but stuck at build time - there's some issue I cannot overcome myself. On the good side, guys at Marlin offered to have a look at E3S1 fork which has many nice thing, has refactored code base and might be ok, but it would require us to properly incorporate e3v3se features into it. So, it's possible, but there's a lot of work ahead.

demxp, 2025-05-02:
A continuation of the story with the latest version of the motherboard has appeared on Reddit. They posted a Klipper, but maybe the details will help. In particular, the offset, maybe. Here is a quote from there.

https://www.reddit.com/r/Ender3V3SE/comments/1k2dptx/comment/mpwtqo4/

`
Enter Klipper with the command "cd ~/Klipper"

Run the "make menuconfig" command

Select these settings:

Copy code-Microcontroller architecture: STMicroelectronics STM32

Processor model: STM32F401

Bootloader offset: 64KiB

Communication interface: Serial (in USART1 PA10/PA09)

Press Q to exit make menuconfig and Y to save

Run the "make" command

Use WinSCP to connect via SSH to the Raspberry Pi, go to the "klipper" folder and select the Klipper .bin file. Or just move the .bin file to a USB. How do you want to get the .bin file in a place where you want to format a pendrive and put the .bin in a folder called "STM32F4_UPDATE"

Copy the .bin to the desktop

Format your SD card in FAT32. If it is larger than 8 GB, create a smaller main partition, less than 8 GB.

Create a folder on the card and call it "STM32F4_UPDATE"

Copy the klipper.bin file to that folder and eject the SD card, don't just take it out.
`

eduard-sukharev, 2025-05-02:
Thanks for heads up. I've asked at Discord for my errors and they advised me to have a look at [this close-to-upstream repo](https://github.com/ThomasToka/Marlin). I'll try to backport their file structure and organize our files in similar fashion so that we could make printer specific changes localized and discreet as much as possible in future.

> Are these boards available to purchase?

I haven't found them by the `CR4NS200141C14` hardware reference, but boards for Ender 3 S1 are freely available in general. It could be nice to have one for sure, but we have to first solve the basic build issues.

My current first two attempts are in these [two](https://github.com/eduard-sukharev/Ender-3V3-SE-Octoprint/tree/add_STM32F401_support_granular) [branches](https://github.com/eduard-sukharev/Ender-3V3-SE-Octoprint/tree/add_STM32F401_support). They're mostly the same and fail with pin macro unfolding for `#define P14 29`.

navaismo, 2025-05-02:
> My current first two attempts are in these [two](https://github.com/eduard-sukharev/Ender-3V3-SE-Octoprint/tree/add_STM32F401_support_granular) [...]

The file: stm32f4/pins_CREALITY_V3_F401RET6.h which is missing after cloning your branch is a copy of: pins_CREALITY_S1_F401RC.h ?

What if we check the output files that klipper creates after compiling the software for the new board, maybe we can figure it out mappings, bootloaders and so?

eduard-sukharev, 2025-05-03:
> The file: stm32f4/pins_CREALITY_V3_F401RET6.h which is missing after cloning your branch is a copy of: pins_CREALITY_S1_F401RC.h ?

No, it's copied from different repository, probably from ender5s1 (it uses the board with same chip). 

> What if we check the output files that klipper creates after compiling the software for the new board, maybe we can figure it out mappings, bootloaders and so?

That might be useful, but I have no experience with Klipper and have no idea what it generates - my assumption it's some kind of a binary firmware for slave board while the main workload is on the rpi or something. Or is there something else before the slave firmware built that could be useful for us?

navaismo, 2025-05-03:
>

Yes it generates a binary as a result of the compilation but one step before the user needs to configure what to compile and for what platform/chip that's why they run the `make menuconfig` after they select the settings somewhere Klipper creates the necessary header files with the expected configuration so потом I'll try to confirm the theory (4am here need to sleep a bit).

KaPcTaPk, 2025-05-03:
I had the same firmware 1.0.9_F401,
but I was inattentive.
I tried to install your firmware through the STM32F4_UPDATE folder.
Now I always have a blue download screen with no labels.
what should I do?
the printer is not working

navaismo, 2025-05-03:
> I had the same firmware 1.0.9_F401,
> but I was inattentive.
> I tried to install your firmware through the STM32F4_UPDATE folder.
> Now I always have a blue download screen with no labels.
> what should I do?
> the printer is not working

Uh you shouldn't did that. Different chip, different settings.
In the hopes that the bootloader isn't affect try to flash again the 1.0.9_F401

KaPcTaPk, 2025-05-03:
> > У меня была такая же прошивка 1.0.9_F401, но я был невнимателен. Я попытался установить вашу прошивку через папку STM32F4_UPDATE. Теперь у меня всегда отображается синий экран загрузки без надписей. Что мне делать? принтер не работает
> 
> Э-э, вам не следовало этого делать. Другой чип, другие настройки. В надежде, что загрузчик не повлияет на работу, попробуйте снова прошить 1.0.9_F401

Unfortunately, there is no old firmware on the official website, and I do not know where to get it, the latest version is only 1.0.6.

navaismo, 2025-05-03:
> Unfortunately, there is no old firmware on the official website, and I do not know where to get it, the latest version is only 1.0.6.

попробуйте с этим

[Archive.zip](https://github.com/user-attachments/files/20023867/Archive.zip)

KaPcTaPk, 2025-05-03:
> > > > У меня была такая же прошивка 1.0.9_F401, но я был невнимателен. Я попытался установить вашу прошивку через папку STM32F4_UPDATE. Теперь у меня всегда отображается синий экран загрузки без надписей. Что мне делать? принтер не работает
> > > 
> > > 
> > > Э-э, вам не следовало этого делать. Другой чип, другие настройки. В надежде, что загрузчик не повлияет на работу, попробуйте снова прошить 1.0.9_F401
> > 
> > 
> > К сожалению, на официальном сайте нет старой прошивки, и я не знаю, где её взять. Последняя версия — 1.0.6.
> 
> попробуйте с этим
> 
> [Archive.zip](https://github.com/user-attachments/files/20023867/Archive.zip)

Everything worked, thanks a lot!

navaismo, 2025-05-05:
I was looking at the string in the BIN file that was revoked by Creality and found this:

```diff
+V1.0.9_F401
+Last Updated: 2022-11-28 | Author: Creality
+Compiled: Nov 15 2024

+FIRMWARE_NAME:Marlin V1.0.9_F401 (Nov 15 2024 16:32:26) 
+SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin 
PROTOCOL_VERSION:1.0 
MACHINE_TYPE:Ender-3 V3 SE 
EXTRUDER_COUNT:1 
UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

+C:\Users\103630\.platformio\packages\framework-arduinoststm32\variants\MARLIN_CREALITY_STM32F401RE\variant.cpp
C:\Users\103630\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.cpp
C:\Users\103630\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareTimer.cpp
C:\Users\103630\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring_analog.c
C:\Users\103630\.platformio\packages\framework-arduinoststm32\cores\arduino\wiring_digital.c
C:\Users\103630\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32\pinmap.c

``` 

But cannot find the variant MARLIN_CREALITY_STM32F401RE even using the E5-S1 configuration, about Klipper I got a bunch of definitions but hard to translate to Marlin.

navaismo, 2025-05-05:
> but we have to first solve the basic build issues.

I have created a branch from our main branch repo not from Marlin upstream, copied the original V3SE pin configuration and renamed it for the F401, added the variants and the board of F401 from Ender 5 S1 repo and compiled successfully in the branch https://github.com/navaismo/Ender-3V3-SE/tree/SMT32F401

Obviously it was just a test to see if can be compiled, now we need the correct PIN mapping for the new board and edit the `Marlin/src/pins/stm32f4/pins_CREALITY_F401.h` so what do you think is the best approach: Using the PINS from Ender 5 S1 or trying to get the maps from Klipper(since people reports it is working)?

navaismo, 2025-05-05:
> I think we can get e5s1 and e3s1 pin mappings. I'm sure they are very similar. First tests will be very... guessing. But i'm sure once you've made it all build then we can advance pretty fast

From Klipper conf file for E5S1:

```conf
# Creality Ender 5 S1 (HW version: CR4NS200141C13)
#
# printer_size: 220x220x280
# To use this config, during "make menuconfig" select the STM32F401
# with a "64KiB bootloader" and serial (on USART1 PA10/PA9)
# communication.
#
# Flash this firmware by creating a directory named "STM32F4_UPDATE"
# on an SD card, copying the "out/klipper.bin" to it and then turn
# on the printer with the card inserted. The firmware filename must
# end in ".bin" and must not match the last filename that was flashed.
#
# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PC2
dir_pin: !PB9
enable_pin: !PC3
rotation_distance: 40
microsteps: 16
endstop_pin: !PA5
position_endstop: 220
position_max: 222
homing_speed: 80

[stepper_y]
step_pin: PB8
dir_pin: !PB7
enable_pin: !PC3
rotation_distance: 40
microsteps: 16
endstop_pin: !PA6
position_endstop: 220
position_max: 220
homing_speed: 80

[stepper_z]
step_pin: PB6
dir_pin: PB5
enable_pin: !PC3
rotation_distance: 8
microsteps: 16
endstop_pin: probe:z_virtual_endstop
position_max: 280
homing_speed: 20
second_homing_speed: 1
homing_retract_dist: 2.0

[extruder]
step_pin: PB4
dir_pin: PB3
enable_pin: !PC3
rotation_distance: 7.5
microsteps: 16
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid  # tuned for stock hardware with 210 degree Celsius target
pid_kp: 20.749
pid_ki: 1.064
pid_kd: 101.153
min_temp: 0
max_temp: 305
[... 667 lines cut ...]
  #define Y_SERIAL_TX_PIN                     PB13
  #define Y_SERIAL_RX_PIN                     PB13

  //#define Z_SERIAL_TX_PIN                   X_SERIAL_TX_PIN
  //#define Z_SERIAL_RX_PIN                   X_SERIAL_RX_PIN
  #define Z_SERIAL_TX_PIN                     PB14
  #define Z_SERIAL_RX_PIN                     PB14

  //#define E0_SERIAL_TX_PIN                  X_SERIAL_TX_PIN
  //#define E0_SERIAL_RX_PIN                  X_SERIAL_RX_PIN
  #define E0_SERIAL_TX_PIN                    PB15
  #define E0_SERIAL_RX_PIN                    PB15

  #define X_DIAG_PIN                          PB10
  #define Y_DIAG_PIN                          PB11
  //#define Z_DIAG_PIN                        PB14
  //#define E0_DIAG_PIN                       PB15

  // Reduce baud rate to improve software serial reliability
  #define TMC_BAUD_RATE                       19200 // 9600
#endif

```

I'm trying to dig into Klippers confs no luck so far.

navaismo, 2025-05-07:
Asked Creality about when they are going to release the code for the new board and they politely told me basically I'm dumb:

[image]

:@ 😡

eduard-sukharev, 2025-05-07:
Send them screenshot for CN14 hardware screen and ask them where to get the firmware for it. And then ask where is the source for it.

Anyways, @CrealityTech is as bad at open source as Makerbase, now we just know it.

AlexCuke, 2025-05-09:
> Send them screenshot for CN14 hardware screen and ask them where to get the firmware for it. And then ask where is the source for it.
> 
> Anyways, [@CrealityTech](https://github.com/CrealityTech) is as bad at open source as Makerbase, now we just know it.
I wrote to support creality and received the following response
"If the firmware currently available on our official website is not suitable for your Ender-3 V3 SE, and you've already attached a photo of the motherboard, we will need to verify the specific board version to ensure compatibility.

Please allow us a moment to review the photo. In the meantime, make sure your request includes:

The exact board version number (e.g., V4.2.2, V4.2.7, or V4.3.x).

Whether you are using a stock display or a touch screen.

Any modifications you’ve made to the printer (e.g., CR Touch, new hotend, etc.).

Once confirmed, we will send you a direct link to the correct firmware."

demxp, 2025-05-09:
Hm. This is strange. I received an email about a new comment in the subject, but the comment itself is missing. The letter is given below. Judging by the text, no one plans to disclose the source code of the firmware, but they want to help, so to speak, privately.

But the finished firmware is not needed, the source code for the new hardware is required in order to configure this custom firmware for it. The official one is already available. I don't understand.

`Re: [navaismo/Ender-3V3-SE] Compatibility with CR4NS200320C14 hardware (STM32F401RET6) (Issue #96)
AlexCuke left a comment (navaismo/Ender-3V3-SE#96)
Send them screenshot for CN14 hardware screen and ask them where to get the firmware for it. And then ask where is the source for it.

Anyways, @CrealityTech is as bad at open source as Makerbase, now we just know it.
I wrote to support creality and received the following response
"If the firmware currently available on our official website is not suitable for your Ender-3 V3 SE, and you've already attached a photo of the motherboard, we will need to verify the specific board version to ensure compatibility.

Please allow us a moment to review the photo. In the meantime, make sure your request includes:

The exact board version number (e.g., V4.2.2, V4.2.7, or V4.3.x).

Whether you are using a stock display or a touch screen.

Any modifications you’ve made to the printer (e.g., CR Touch, new hotend, etc.).

Once confirmed, we will send you a direct link to the correct firmware."`

[image]

navaismo, 2025-05-09:
> Hm. This is strange. I received an email about a new comment in the subject, but the comment itself is missing

I saw it here too, seems like the author deleted the message.

AlexCuke, 2025-05-09:
> > Hm. This is strange. I received an email about a new comment in the subject, but the comment itself is missing
> 
> I saw it here too, seems like the author deleted the message.
I deleted the message because I realized that this is an interim response from technical support. Now I continue to communicate with technical support. When I receive an answer, I will publish

navaismo, 2025-05-12:
> Send them screenshot for CN14 hardware screen and ask them where to get the firmware for it. And then ask where is the source for it.

I did send the screenshot of the Board with the SMTF401 and clearly stating that the repo is not for that chip and this was their answer.

I feel mad and they made me feel stupid 😡😡😡😡

[image]

eduard-sukharev, 2025-05-12:
Can we send them an LCD error screen showing that we cannot flash the "opensource version" to it? Do we have it? they are definitely playing dumb to escape any responsibility whatsoever

KaPcTaPk, 2025-05-23:
[image]
Oh, the new firmware for the F401 has been posted on the creality website.
In case someone missed it.
I will support the development at least in this way))))))
https://www.crealitycloud.com/downloads/firmware/ender-series

navaismo, 2025-05-23:
We need the source code, we have already the binaries :(

KaPcTaPk, 2025-05-26:
https://github.com/CrealityOfficial/Ender-3V3-SE/issues/13
A question about the source code was recently asked here.
Maybe if there is a big discussion with a lot of people, Creality will pay attention to it))))

eduard-sukharev, 2025-05-26:
I'm currently working on upstream Marlin migration to our printer. It already has most of the needed code for both revisions of boards, but the screen is very different and is a general PITA, so it's coming along slowly. I hope we'll be able to switch to it at some point.

AIc1n, 2025-05-26:
hi. I have a printer on the same board, and I can use a multimeter to find where the screen is connected.

AIc1n, 2025-05-26:
so here is the stm32f401ret6 contact list relative to the screen:
PA12 -> (A) encoder screen left
PA11 -> (B) encoder screen right
PB1 -> (ENT) button encoder screen
PB0 -> (BEEP) beeper
PA2 -> (TX) screen
PA3 -> (RX) screen
the last two can be inverted, as indicated on the screen board, I hope this will help in the development of the project.

navaismo, 2025-05-26:
> so here is the stm32f401ret6 contact list relative to the screen: PA12 -> (A) encoder screen left PA11 -> (B) encoder screen right PB1 -> (ENT) button encoder screen PB0 -> (BEEP) beeper PA2 -> (TX) [...]

Awesome! let me check in the pin definition and try.
Thanks for the help.

AIc1n, 2025-05-26:
continue to test

navaismo, 2025-05-26:
> continue to test

Awesome, thx for the help.

AIc1n, 2025-05-26:
It's beneficial to me too, to some extent, so thank you too.

navaismo, 2025-05-26:
> Well, the printer prints and works, unfortunately, there is no way to test with octaprint right now. If there are any problems, I will write

Excellent, the Octoprint part is optional. If the normal behaviour is working fine then octoprint integration will be fine.
So I'll add the latest code and release the latest version for this Hardware too.

I really appreciate your great help to move forward this issue.

wzd2002, 2025-06-06:
I admire this colossal work that you have done to launch this firmware!
Tell me, do you have a solution that can connect creality nebula pad to the printer Ender 3 v3 SE with the board cr4ns200320c14?

navaismo, 2025-06-06:
> do you have a solution that can connect creality nebula pad to the printer Ender 3 v3 SE with the board cr4ns200320c14?

This comment https://github.com/navaismo/Ender-3V3-SE/issues/96#issuecomment-2848093239 describe how to install klipper, does that helps to connect the Nebula?

navaismo, 2025-06-06:
To make it work you need to read carefully the instructions on how to compile the firmware in the comments mentioned above, and also seems that you need to root the Nebula PAD too according a quick google search.

So, if you have a hard time compiling or applying those changes is going to be difficult to achieve it since all steps require some techy knowledge.

navaismo, 2025-06-06:
> At the moment I have installed root firmware on nebula pad. But apparently, it is impossible to compile firmware on Nebula. For now I have stopped at the fact that it is necessary to install klipper somewhere to compile firmware there. Does it matter where it will be compiled? On nebula pad or somewhere else?

No doesnt matter, you can use a RPI.

navaismo, 2025-06-06:
I've Cloned the Klipper repo, followed the instructions of https://github.com/navaismo/Ender-3V3-SE/issues/96#issuecomment-2848093239  and it produced the below.

[klipper.bin.zip](https://github.com/user-attachments/files/20632193/klipper.bin.zip)

Check if works for you, Not guarentee since i dont use klipper I just followed the instructions.
Remember to unzip it and copy it inside the SMT32_UPDATE folder.

wzd2002, 2025-06-07:
navaismo, thank you! Perhaps you meant the SMT32F4_UPDATE folder? Just in case, I tried both options.
Unfortunately, this did not help. If you use root firmware, then everything ends with the message "Anomaly in communication with the machine's motherboard"
If you use firmware from the creality website - NEBULA_ota_img_V1.1.0.29 ( In versions below, the same thing)

I tried to select the Ender-3 S1 printer with the STM32F401 controller when setting up the Nebula Pad - I got the same error

It's probably not the printer board firmware, but the Nebula Pad itself.

[image]
[image]
[image]
[image]

wzd2002, 2025-06-07:
yes, I tried. Maybe even during this time crealityy will release some update
Thanks for trying.
In principle, your firmware with Marlin works fine, but as they say "Дурная голова ногам покою не дает" ))