navaismo/Ender-3V3-SE #97: Control / Printer Statistics

andersborgabiro, 2025-04-24:
The numbers are cut off to 3 digits, so 1xxx (etc) becomes 1xx.

andersborgabiro, 2025-04-24:
[image]

andersborgabiro, 2025-04-24:
DWIN_Draw_IntValue(true, true, 0, font6x12, Color_White, Color_Bg_Black, 3, VALUERANGE_X, MBASE(1) + 3 , totalPrints);

navaismo, 2025-04-24:
Good catch! Another fix to the list. 
Do you guys think is safe making the length for 6 digits?

andersborgabiro, 2025-04-24:
Close call space-wise.

If any user hits that amount they should get an award (another SE) :).

navaismo, 2025-05-02:
Fix implemented in the [latest release 1.0.9.8](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.9.0.8), let us know if now looks better.

closed 2025-05-02 by navaismo

andersborgabiro, 2025-05-02:
I see 4 digits now 👍.