navaismo/Ender-3V3-SE #98: Enable UART on "LCD" port

1StNiRoM, 2025-04-25:
If we could enable UART on the LCD port (near USB port).

With this we should be able to power a Raspberry Pi and send commands to the printer internally without the need for an externally mounted and powered Raspberry Pi.

eduard-sukharev, 2025-04-25:
That is an interesting take. Did you actually try to connect to exp1 to get a UART?

currently mcu already talks to lcd with plain serial tx/rx communication, which is UART or USART if I get it right. I'm not sure if it's mutual and mcu can accepts command from the other side (ie lcd or rpi), so you're free to try it as is. the baudrate is 115200 always

eduard-sukharev, 2025-04-28:
@1StNiRoM if you need more info on which pins to use for TX/RX and power — take apart the screen case and have a look at EXP1 connector pinout there. Also, have a look at [this great mriscoc's writeup on reverse engineering this screen](https://jpcurti.github.io/E3V3SE_display_klipper/tjc3224_reverse_engineering.html) — it has pics describing where and how to connect, I'm sure you can try something like that.

closed 2025-04-28 by eduard-sukharev