navaismo/Ender-3V3-SE #100: Beep when preheat done

andersborgabiro, 2025-04-29:
This is not a bug report but a feature request:

As I have a tendency to forget to check when Preheating is completed, and generally, a beep once the preheat (PLA etc) has reached the temperature level would clearly indicate it's time to take action.

navaismo, 2025-04-29:
Sounds doable. 3 beeps(beep-beep-beeeeeeep) would be enough?

navaismo, 2025-04-29:
> Just to remind you: what should happen if screen beeper is muted? should it still beep after preheat is done or should it be quiet?

I think like the M600 its an alert sound so we can override the Mute of LCD.

> Also, should it beep only after preheat from menu, or should it also beep after the initial preheat in G-Code print from SD/OctoPrint?

As I understood is only for the preheat routines not for the print file isn't

andersborgabiro, 2025-04-29:
> As I understood is only for the preheat routines not for the print file isn't

That was my thought. That's when action is required.

Heating during printing is a step in the process and requires no intervention.

Dan263737, 2025-04-29:
> Just to remind you: what should happen if screen beeper is muted? should it still beep after preheat is done or should it be quiet? Also, should it beep only after preheat from menu, or should it also beep after the initial preheat in G-Code print from SD/OctoPrint?

I think it should be individual setting.

eduard-sukharev, 2025-04-29:
Agreed. Moreover, it probably should be grouped with "Mute beeper" setting into its own subgroup.

navaismo, 2025-04-29:
I would like to avoid making more Menus not because is a pain making menus for the LCD along with the HMI control but mostly because with the incoming release new fixes and features, we are already reaching 50% of resources for RAM and Flash. Which led us into a red zone. When I was doing the Gcode Preview branch consuming more than 60% of the resources starting to give problems while printing because there will be not much space for the HAL programm to buffer the commands.

Adding this feature along the rest of the ixes we are right now with:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.0% (used 32800 bytes from 65536 bytes)
Flash: [=====     ]  48.5% (used 254284 bytes from 524288 bytes)
```

navaismo, 2025-05-02:
Implemented in the latest r[elease 1.0.9.8](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.9.0.8).

closed 2025-05-02 by navaismo

andersborgabiro, 2025-05-02:
It works fine 🙂👍.

(I forgot about preheating as recently as yesterday; updated firmware today)

Dan263737, 2025-05-02:
Works good, but I like when the printery is silent and beeping disturbes me. Can you please make a setting for turning it off? Thanks

andersborgabiro, 2025-05-04:
It only beeps when you preheat manually and the temperature has reached the setting. The whole point is to avoid the printer heating for no reason and also to avoid wasted time.

Ray2Jerry, 2025-05-07:
I would love an option to turn this off like other beeps. I print late at night and that alarm is very loud and unecessary if I'm standing next to it. Thank you!

Dan263737, 2025-05-07:
Same thing. Please

navaismo, 2025-05-08:
@eduard-sukharev @Ray2Jerry @Dan263737 

Added the Menu to silence the Preheat Alert in the [Release v1.0.9.8_1b](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.8_1b).

It cost us .1% of Flash

Dan263737, 2025-05-08:
What do you think is the last safe ℅ of flash used?

navaismo, 2025-05-08:
> What do you think is the last safe ℅ of flash used?

50%

We are currently:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.0% (used 32800 bytes from 65536 bytes)
Flash: [=====     ]  48.6% (used 254876 bytes from 524288 bytes)
```

eduard-sukharev, 2025-05-08:
What's the current status of your thumbnail preview fork? what's the ram usage? i saw the reorganized menu and i think it's cool - it could be used with sd printing too.

navaismo, 2025-05-08:
> What's the current status of your thumbnail preview fork? what's the ram usage? 

for the Gcode Preview is worst haha:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.4% (used 33016 bytes from 65536 bytes)
Flash: [=====     ]  49.1% (used 257396 bytes from 524288 bytes)
```

> i saw the reorganized menu and i think it's cool - it could be used with sd printing too.

Which Menu?