navaismo/Ender-3V3-SE #105: CR touch test

Dan263737, 2025-05-07:
### Describe it

Hi, a few days ago I had a CR touch error (it was not caused by the firmware) (I fixed it), it did not want to deploy and I needed to test it by auto-homing.  And that is why I came with an idea- option to manually deploy or retract CR-touch. 
i think this wont be pain to meke this. Sending a video of my friend's ender 3 with this option. Thanks

Dan263737, 2025-05-07:
https://github.com/user-attachments/assets/2bf2c7bc-861b-44c6-8286-1fea3f04bf45

Dan263737, 2025-05-07:
sorry for bad quality video

diegoemy76, 2025-05-07:
I wanted to write it too but you beat me to it, a function like that would be very convenient

eduard-sukharev, 2025-05-07:
This sounds like a nice to have option. Mind, though, that we're at the point when adding new screens menu options for features is prone to degrading printer performance as there's pretty much limited amount of RAM on this device.

navaismo, 2025-05-07:
I just tried it  by sending a SERIAL Command to avoid creating a Menu and worked good but I guess not everyone has connected a Serial terminal to the printer.

> Mind, though, that we're at the point when adding new screens menu options for features is prone to degrading printer performance as there's pretty much limited amount of RAM on this device.

I'm wondering what will be less memory consuming leaving the options floating there like at the end of prepare menu or adding it to the Move Menu at the end too

diegoemy76, 2025-05-07:
> I just tried it by sending a SERIAL Command to avoid creating a Menu and worked good but I guess not everyone has connected a Serial terminal to the printer.
>
>
> I'm wondering what will be less memory [...]

Putting it in the move menu makes more sense though

Dan263737, 2025-05-07:
Looks good

eduard-sukharev, 2025-05-07:
Looks good. Our menus need some re-shuffling, but putting these into Move seem quite appropriate right now. Roll it.

navaismo, 2025-05-07:
Release into [v1.0.9.8_1a](https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.8_1a)

closed 2025-05-07 by navaismo

Wingtiger, 2025-05-07:
I've just thought about overclocking the microcontroller core a bit...

Dan263737, 2025-05-08:
> I've just thought about overclocking the microcontroller core a bit...

If you want to discuss this, open another issue,