navaismo/Ender-3V3-SE #106: Insanely loud after the last release

PythonEatsSQuirreL, 2025-05-09:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I am using FW10979_OctoPrintPageinLCD_150kbps for a couple of months and I am really happy with it. 
I saw there are a few new releases including the one that is showing the print hours. 
I updated to FW10981a_150Kbps, the printer started to move a lot faster on every command, printed hours added and everything else BUT, during the print and leveling the printer is super super loud (which doesn't happen on stock 1.0.6 or FW10979. 
Was the voltage adjusted on the motors in the most recent versions? Is the only possible thing that would explain the horrific sound. 

### Expected behavior

Meanwhile I switched back to FW10979 from FW10981a and everything is silent again.

PS: I'm not using octo, just SD card plug and play.

### Actual behavior

Furthermore, the new version of estimating the print time takes quite a while and it's still off by a lot (2h estimating on a 15 min print). 

### Steps to Reproduce

FW10979_OctoPrintPageinLCD_150kbps and print anything.

### Your Slicer

Orca Slicer

### Host Software

SD Card

### Additional information & file uploads

If it matters, I printed the calibration cube in PETG in order to test (or just the first 2 layers, I had to stop it because it was too brutal.

navaismo, 2025-05-09:
Hi, this is not a bug is a feature request to bump the speed of the printer, which is documented in the Release.
You have two options regarding the speed and sound:

1.- Decrease the Max Acceleration in the motion menu to 2000, and decrease the Max Speed also in Motion Menu to 200 or less for XY and 5 for Z.

2.- Recompile the Firmware to roll back the acceleration values:
Marlin/Configuration.h
```diff
// Homing speeds (mm/min)
+#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (8*60) }
-#define HOMING_FEEDRATE_MM_M { (80*60), (80*60), (50*60) }
``` 

Regarding the Time Issue @eduard-sukharev is still looking at it, the way creality is calculating it is not reliable so He's trying to make a fix on that.

PythonEatsSQuirreL, 2025-05-09:
Thank you very much for your answer and also for all your efforts and commitment on these upgrades!

closed 2025-05-09 by PythonEatsSQuirreL

Monkeylogg, 2025-05-10:
Hi, can anybody help me with recompiling this firmware, i downloaded the zip from this github, downloade vs code, installed PlatformIO, i even tried Auto Build Marlin. Even when i dont change anything the compiling fails when i change the acceleration values to stock it still fails. I dont know what to change at this point

eduard-sukharev, 2025-05-10:
> when i dont change anything the compiling fails when i change the acceleration values to stock it still fails

**Create new issue and don't post on closed issue**. Provide at least some logs, elaborate on what you get so that community can help you.