navaismo/Ender-3V3-SE #108: Need help with compiling this firmware

Monkeylogg, 2025-05-10:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I tried downloading this firmware, changing values from issue #106 :
"Recompile the Firmware to roll back the acceleration values:
Marlin/Configuration.h

// Homing speeds (mm/min)
+#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (8*60) }
-#define HOMING_FEEDRATE_MM_M { (80*60), (80*60), (50*60) }"

but when i press build it fails, i tried following like 4 youtube guides. I used VS code and installed PlatformIO,
but no luck, i also tried Auto Build Marlin plugin but it also failed even when i did not change anything.
At this point i dont know what to change i tried to tinker with some settings and looked some help on reddit but it always fails. (I also have no coding knowledge at all and its my first time compiling any firmware).

### Actual behavior

Also when i unpack and open it it says platformio.ini has 12 problems
Here is some logs and a screenshot:

[image]
[1st time build with PlatformIO.txt](https://github.com/user-attachments/files/20139980/1st.time.build.with.PlatformIO.txt)
[2nd time build with PlatformIO.txt](https://github.com/user-attachments/files/20139979/2nd.time.build.with.PlatformIO.txt)

### Steps to Reproduce

1. Download ZIP from github from main branch (Code button and download ZIP)
2. Unzip ZIP ( I used winrar) and open it in VS code
3. Install PlatformIO and Auto Build Marlin
4. Try to compile with either of those
5. It fails
6. Try changing acceleration values, it still fails

### Your Slicer

Orca Slicer

### Host Software

SD Card

navaismo, 2025-05-10:
I use Linux to compile so maybe the linker was cached.
Do a cleannup of the project and then compile again.

navaismo, 2025-05-10:
Also, in your folder path do not use special characters or spaces. So move your project to a standard path like:

C:\tmp\e3v3se\

And try compiling from there.

Monkeylogg, 2025-05-10:
well that seemed to fix the building issue but it still highlights the problems but im just gonna ignore it

[image]

[image]

Monkeylogg, 2025-05-10:
Now im gonna change values and rebuild and reflash

Monkeylogg, 2025-05-10:
Okay so im changing these with Auto build marlin because its more clearer than text editor:

[image]

[image]

[image]

And it worked :) but when i tried to turn off power loss recovery in firmware it would not compile, it kinda does not matter because i set it in orca slicer's gcode

[image]

Monkeylogg, 2025-05-10:
Okay so the problem still persists I recorded 2 videos. First is Homing by menu, its much quieter but after printer hits X and Y endstops and its going to check Z height its making wierd noise at stepper motors (its quieter so my phone didnt pick it up) (I cant upload videos to github most likely too much size im gonna host it on other website). Homing before printing is as loud as precompiled firmware i downloaded for the first time, i used the same gcode from orca slicer on stock and this firmware and its definitely not the gcode, i have turned on emiting limits to gcode in orca so it should not be a problem.

[image]

here are the videos:

https://limewire.com/d/T16CX#r7hWfdTzLA

navaismo, 2025-05-10:
Compare the values with the creality stock firmware and setup according your needs.

Since you manage to compile and this issue is not related anymore with the firmware I'm going to close the issue.

If you need further help regarding stock settings you can open an issue on creality GitHub or contact their support.

closed 2025-05-10 by navaismo