navaismo/Ender-3V3-SE #119: Weird beeping and scraping on printer

ademharacic3421, 2025-06-04:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I was homing, but when i home the x axis motor overheats and starts beeping while moving and grinds a little. The y axis too.

### Expected behavior

Expected to home without problems.

### Actual behavior

Scraped and beeped (no physical damage tho)

### Your Slicer

Creality Print

### Host Software

SD Card

navaismo, 2025-06-04:
- After the change of firmware did you exectud a new bed level mesh regarding the scrapping?
- Heating of motors is totally new are the motors being modified or not the originals? 
- Are you using a PLA/PETG enclosure?

ademharacic3421, 2025-06-05:
No, the nozzle didnt scrape the bed the x and y axis scrape when moving, like the rods and stuff
The motors heat only to 55c so i think its normal
and i do not have a enclosure
Btw i tried again with stock it worked so im gonna try take the home command from the old firmware and paste into your beautiful firmware :)