navaismo/Ender-3V3-SE #120: Support for Biqu MicroProbe

0pc56, 2025-06-05:
### Describe it

Hello. I have a Biqu MicroProbe V2 and have been trying to get it to work but no matter how much I modified the configuration.h the probe does not extend as it should. more information can be found [here](https://github.com/bigtreetech/MicroProbe)
the probe detection works but it's flipped as tested when i pulled the probe.

navaismo, 2025-06-05:
Have you tried the Deploy and Stow options within Move Menu to validate the probe works?

Also you can try to disable the High Speed configuration for the default CR touch as explained in this PR: https://github.com/navaismo/Ender-3V3-SE/pull/19

0pc56, 2025-06-05:
tried the deploy and stow, doesnt work as intended. turned off bltouch_hs_mode aswell.

navaismo, 2025-06-05:
Seems so.

0pc56, 2025-06-05:
ill make the changes later and update accordingly

navaismo, 2025-07-24:
Close due to inactivity

closed 2025-07-24 by navaismo